paulh002/sdrberry #4: Project development

UN7RX, 2022-04-20:
Hello! A very beautiful project. It is surprising that little is known about him. Will there be HPSDR support? And support for external buttons, such as in piHPSDR

paulh002, 2022-04-20:
Thanks, it could support HPSDR protocol, Radioberry is based on hpsdr. But currenlty there is no SoapySDR driver for devices like hermeslight and I don't own such a device. 
But it would be not to difficult to add HPSDR support direct once I have a device I will  give it a go.

UN7RX, 2022-04-20:
Thanks a lot for your answer, Paul!
I think  can try using this console
https://github.com/VU2DLE/Radioberry_Console
What do you think about this?
I have been looking for a program for the Raspberry Pi for a long time, of course everyone knows piHPSDR from g0grx. But it has a very unattractive interface.
Ram, VU3RDD made it prettier.    https://www.youtube.com/watch?v=0pO9re2Jr84
But your version looks great!

paulh002, 2022-04-20:
Looks great, I think you can use VU2DLE console. Should not be to difficult to support it. If you check my repositories for VFO ESP32 and ESPControler you can see examples of the controler I made it uses small touchscreen instead of buttons. The SDR tranceiver is it your own design or variation on Hermes Light2? Look very interesting. Problem with radioberry is the availability of the FPGA. So if there are alternatives would be nice. You can check a demo at my youtube channel https://www.youtube.com/watch?v=ycU_IUmV8j4
[image]

UN7RX, 2022-04-20:
Interesting video. But I didn’t understand if there is a noise suppressor and impulse noise? NB and NR?
And another moment. Nothing is clear on the spectrum analyzer. It is absolutely chaotic, the signal is not visible on it.

There is such a project, from Laurence Barker, the "ODIN" console for PowerSDR. It also has a separate display for buttons and an extended menu.
https://github.com/laurencebarker/odin-SDR-console

UN7RX, 2022-04-21:
Very sorry. Without such functions, the transceiver loses a lot. Digital signal processing, noise and interference suppression for DDC DUC is probably a must-have feature.

UN7RX, 2022-04-24:
I tried installing SDRBerry today. Unfortunately, it doesn't work yet. The first part goes well. The last command takes a very long time to complete. As if I installed half of the Debian repositories))))

./sdrberry/install/install.sh

But then there are problems, here is the block of code.

```
git clone https://github.com/paulh002/sdrberry  
cd sdrberry  
mkdir build  
chmod +x ../do_cmake.sh  
../do_cmake.sh  
make
sudo make install  
```

There are problems when trying to compile
[image]
Please note that the code in this form ../do_cmake.sh does not work for me at all. Only do_cmake.sh or ./do_cmake.sh
Either way, it throws an error. Until I got further..

paulh002, 2022-04-24:
Error in the readme page. You have to cd info build directory.
Also note sdrberry uses framebuffer to write to display. You have to switch of xwindows desktop before starting sdrberry.
Just user raspi-config and in System menu switch to console only.

Also I did a lot of updates in dev branch, but need to update installation script for it.

UN7RX, 2022-04-24:
Thanks for the reply Paul!
I think that the installation option that is now is designed for a well-prepared Linux user. And knowledgeable Raspbian. Unfortunately, I'm not very good at this yet. Therefore, I will not take your time, maybe in time your beautiful software will be installed even by beginners.
Installing piHPSDR is very easy and automatic. I'll put either him or Thetis with a hardware console.
Good luck, thanks!

paulh002, 2022-04-24:
You are right, installation is complicated. There are several way's to make it more easy, like using docker or put all libraries directly in the git repository. Have to thinkt about it. Any way thanks for the interest. 
But the SDR receiver you build is there any information about it? Looks very good. 
Any way thanks and see you later 73.

UN7RX, 2022-04-24:
This is a fairly simple device, AD6645 + DAC904, audio codec on WM8731 (but it is not required, it also works fine without it), FPGA EP4CE22E22C8N. The module with FPGA is used ready - so there are fewer problems with the layout of the board. I have a module with EP4CE22F17C8 which is even smaller but has a lot more ports. This allows very good development of the scheme.
This [idea is taken as a basis](http://www.cqham.ru/forum/showthread.php?35923-DDC-DUC-%F2%F0%E0%ED%F1%E8%E2%E5%F0-%F1-%E8%ED%F2%E5%F0%F4%E5%E9%F1%EE%EC-Ethernet-%E8%E7-%EC%EE%E4%F3%EB%E5%E9-%F1-%C0%EB%E8). 

Then you can do anything - two ADCs, any peripherals, and so on. Sources are open.

UN7RX, 2022-04-29:
Paul, will you continue to develop your program? Or will it not be fast?

paulh002, 2022-04-29:
Yes, I am still developing the program. There are some items I am working on currently. 1) Improve soapyradioberry driver transmit still has some issues. 2) support for hermes light2. 3) FT8 transmit. 4) GUI improvement. But I also want to improve installation. Is to complicated currently.

UN7RX, 2022-04-30:
Thank you! The most pleasant interface for a 7 inch display for a software transceiver. If you manage to finalize everything, it will be the most popular! Maybe I still have time to put it)))) Good luck to you!

closed 2022-05-01 by paulh002