paulh002/sdrberry #9: libraries

mpvano, 2022-07-30:
Paul, this is a great project, but I'm having no luck figuring out its dependencies.

Do you have a list of the actual names of these library packages and some hints about where to find them?

thanks,

M

paulh002, 2022-07-30:
Hi,

In the install directory there is a script install1.sh this will install all dependencies in the raspberry pi. I assume you will use your own hat (other install files are for radioberry hat or pluto)

copy install1.sh to the pi homedir and execute chmod +x install1.sh them execute ./install1.sh This is the most easy way. I will also check today if this method is not broken.

For the sd card best to install pi os bullseye with desktop, update and configure pi os and then use raspi-config to switch to commandline only interface. If you install commandline only a lot of dependencies are missing.  If you have questions let me know. 

73,
Paul

mpvano, 2022-07-30:
I'll give that a try!

closed 2023-04-17 by paulh002