paulh002/sdrberry #17: Cannot install due to wsjtx_lib.h missing

TechMindsYT, 2023-07-24:
Hi, there. This looks like a really good project and I would like to try it.

I am having this issue where using the install script:
`/home/pi/sdrberry/sdrberry.cpp:20:10: fatal error: wsjtx_lib.h: No such file or directory
   20 | #include "wsjtx_lib.h"
`
I could not find any reference on how to install this dependancy?

paulh002, 2023-07-24:
The cmake file should automaticly download the wsjtx library. I will check it. But you can also clone https://github.com/paulh002/wsjtx_lib separatly and build this library and do a make install. See the readme of the library.
Any way will test it on a clean pi to see if the cmake file is complete

TechMindsYT, 2023-07-24:
Thanks Paul, I found the repo on your github and installed manually. Seems to run now but not entirely sure it fully installed as it rebooted itself. Just working out how to get audio out and should be good to go! looks nice!

paulh002, 2023-07-24:
For audio the best is to use a USB audio adapter, and switch all audio off of the raspberry pi. If you run sdrberry from the commandline be sure to pipe the console to a file "sdrberry > /dev/null 2>&1"  or install the crontab file which is in the install directory. Then you have auto startup.
If you have any questions you can just ask
73 Paul

paulh002, 2023-07-25:
Hi,

paulh002, 2023-07-25:
Hi, I have updated the install scripts and cmake. Also I upgraded some libraries. Seems the latest version of bullseye made some changes. Best to reinstall the project on a fresh bullseye 64 bit in CLI mode and use the install script to clone the project. For radioberry you can use wget https://raw.githubusercontent.com/paulh002/sdrberry/master/install/install-radioberry.sh there are also scripts for pluto, and other or use the common script wget https://raw.githubusercontent.com/paulh002/sdrberry/master/install/install.sh

closed 2023-09-28 by paulh002