paulh002/sdrberry #19: Board support

RichNeese, 2023-08-26:
do you plan to support the radio berry ?[
https://www.amazon.com/AURSINC-SDR-Radioberry-Raspberry-Broadband/dp/B0C2PTW8XW/ ? 
I am lookign at this unit to build a radio

paulh002, 2023-08-26:
Yes, radioberry is supported because it has a soapysdr driver, you can use
the install_radioberry.sh file to install on the pi

RichNeese, 2023-08-26:
Is there a more direct route to ask questions about your software ? I wanted to know more information and if it can ruin a xterm vs making it standalone?

I would like to see it have js8 call and other digital modes in a menu . So that it soul be a field digital radio

paulh002, 2023-08-26:
You can use github discussion, I opened it on the project.
js8 I don't have implemented yet, only ft8 and ft4/ However, the library I
use does support js8 so it is not difficult to add.
You are free to open a pull request to add it.

RichNeese, 2023-08-26:
I am just starting to get into coding… so I have to learn

paulh002, 2023-08-26:
No problem, all input is much appreciated.  I did not understand the question of xterm, but I think it is about remote access. Sdrberry does not use Xwindows. It writes directly into the framebuffer, that's why the pi has to start in cli mode (text only). My goal was to build an standalone tranceiver without the use of a PC. How ever sdrberry can be used remotely. I use for this framebuffer-vncserver (https://github.com/paulh002/framebuffer-vncserver), you can acces it using a vnc viewer. I am also thinking of a webinterface. I will have a look at js8 but the library code is directly taken from wsjtx project which is written in Fortran to adapt it takes a bit time.
73
Paul

RichNeese, 2023-08-26:
Well I want to build a radio but I want to be able to use on screen js8/ft8/fldigi/sstv and packet apps. And most of them need x
But its all good

closed 2023-08-30 by paulh002