paulh002/sdrberry #22: questions ?

RichNeese, 2023-08-28:
Why is it you build sopaysdr insted of installing the pkg ?

I am working to follow what your doing but its a long trail

paulh002, 2023-08-28:
The python packages are not necessary, older SoapySDR builds needed them.
I build SoapySDR to always have a matching version, packages can be older.

RichNeese, 2023-08-28:
Do I use the install_hackrf.sh ? I missed something

paulh002, 2023-08-28:
The only difference between the install scripts is that they install the
SoaySDR driver for the device.
You can use the one that matches the device, or manual clone the SoapySDR
driver which you want to use.
You can also install multiple SoapySDR drivers if you have multiple devices.
For radioberry use the install-radioberry.sh

closed 2023-08-29 by paulh002