paulh002/sdrberry #23: it installed it.

RichNeese, 2023-08-28:
/root/sdrberry/build/_deps/wsjtx_lib-src/wsjtx_lib.cpp:6:10: fatal error: fftw3.h: No such file or directory
    6 | #include <fftw3.h>
      |          ^~~~~~~~~
compilation terminated.
make[2]: *** [_deps/wsjtx_lib-build/CMakeFiles/wsjtx_lib.dir/build.make:76: _deps/wsjtx_lib-build/CMakeFiles/wsjtx_lib.dir/wsjtx_lib.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:227: _deps/wsjtx_lib-build/CMakeFiles/wsjtx_lib.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
[  0%] Building CXX object _deps/wsjtx_lib-build/CMakeFiles/wsjtx_lib.dir/wsjtx_lib.cpp.o
/root/sdrberry/build/_deps/wsjtx_lib-src/wsjtx_lib.cpp:6:10: fatal error: fftw3.h: No such file or directory
    6 | #include <fftw3.h>
      |          ^~~~~~~~~
compilation terminated.
make[2]: *** [_deps/wsjtx_lib-build/CMakeFiles/wsjtx_lib.dir/build.make:76: _deps/wsjtx_lib-build/CMakeFiles/wsjtx_lib.dir/wsjtx_lib.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:227: _deps/wsjtx_lib-build/CMakeFiles/wsjtx_lib.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

RichNeese, 2023-08-28:
there is no fftw3-dev on bullseye
should it not be fftw2-dev

paulh002, 2023-08-28:
No, it has to be fftw3-dev, see https://www.fftw.org/
fftw3-dev is available i just installed it on a fresh OS (64 bit bullseye)

[image: image.png]

paulh002, 2023-08-28:
Did you upgrade bullseye before install??
sudo apt update
sudo apt upgrade

RichNeese, 2023-08-28:
just updated the repos again for the 1th time its there now

RichNeese, 2023-08-28:
yes
I did a full upgrade
and it only showed fftw2 but fftw3 is there now

RichNeese, 2023-08-28:
Package fftw3-dev is a virtual package provided by:
  libfftw3-dev 3.3.8-2 [Not candidate version]

E: Package 'fftw3-dev' has no installation candidate

RichNeese, 2023-08-28:
it has to be libftw3-dev

paulh002, 2023-08-28:
I don't have this issue, just installed it.
Which version of bullseye did you install?? Raspberry Pi OS Lite (64-bit) ?
This version is missing a lot of packages, best to install the full version
and use raspi-config to switch to cli mode

RichNeese, 2023-08-28:
yes I used the light.  version since we are only doing console
but will grab and rebuild

paulh002, 2023-08-28:
Ok, I see. Will change it

RichNeese, 2023-08-28:
so what board are you using to get mic input and audio output from the your hardware
what dac ?

RichNeese, 2023-08-28:
also I have the waveshare 1024x600 it was what I had on hand.
yes its i2c for touch and the driver is installed and I have it working

paulh002, 2023-08-28:
For audio I use a generic USB audio card. It can be anything.
One that works ok is UGREEN audio adapter. But all the cheap USB audio
cards works.

RichNeese, 2023-08-28:
> I don't know if that works correctly, But you can change the size in
> sdrberry.cpp
> In line 47 and 48 is the display size configured.
> You can change it and rebuild (cd in sdrberry/build and de make and sudo
> make install) not necessary to run the whole install script again.
> An other thing to check is the color depth, it is set to 16. Sometimes you
> have to force the display in to 16 colors in /boot/config.txt
> By adding the lines
> max_framebuffers=2
> framebuffer_depth=16

Ok it built and starts but its complaining no sdr found
I am using the hackrf

paulh002, 2023-08-28:
in the GUI setup page you can choose hackrf.
Or in sdrberry_settings.cfg edit the default to hackrf
It has probably radioberry now as default

RichNeese, 2023-08-28:
All I am not getting touchscreen and its not giving me a menu
I also have to fall back to where you said to change display size

paulh002, 2023-08-28:
It is looking for touch screen driver. The name is configured in the cfg
file.cd The standard is touchscreen = "raspberrypi-ts"
I don't know the name for an usb touch screen, but the names of all input
devices are listed during startup

RichNeese, 2023-08-28:
I have the waveshare. https://www.waveshare.com/wiki/7inch_DSI_LCD_(C)

RichNeese, 2023-08-28:
ok working on htis leave here for notes while I fgure this out\

paulh002, 2023-08-28:
install:

apt install input-utils

and issue

lsinput

this will list all your input devices along with some information about
each device

RichNeese, 2023-08-28:
/dev/input/event10
   bustype : BUS_I2C
   vendor  : 0x416
   product : 0x38f
   version : 4192
   name    : "WS Capacitive TouchScreen"
   phys    : "input/ts"
   bits ev : (null) (null) (null)

paulh002, 2023-08-28:
Just send me the list of names, I will check the code and add the name.
The config line is not working. I will change that.
I will check tomorrow. It a bit late here.

RichNeese, 2023-08-28:
ok

RichNeese, 2023-08-28:
need to put in the doc also that it has to be a waveshare 800x600 . until we can get 1024x600 working as a option

paulh002, 2023-08-28:
You can add the code below to evdev.c line 74 (after ptr = strstr(name,
"ft5x06");)
or replace the file evdev.c in lv_drivers\indev\evdev.c

if (ptr == NULL) // Bullseye changed driver name
      ptr = strstr(name, "WS Capacitive TouchScreen");

/**
 * @file evdev.c
 *
 */

/*********************
 *      INCLUDES
 *********************/
#include "evdev.h"
#if USE_EVDEV != 0 || USE_BSD_EVDEV

#include <stdio.h>
#include <unistd.h>
#include <fcntl.h>
#if USE_BSD_EVDEV
#include <dev/evdev/input.h>
#else
#include <linux/input.h>
#endif

/*********************
 *      DEFINES
 *********************/

/**********************
 *      TYPEDEFS
 **********************/

/**********************
 *  STATIC PROTOTYPES
 **********************/
int map(int x, int in_min, int in_max, int out_min, int out_max);

/**********************
 *  STATIC VARIABLES
 **********************/
int evdev_fd;
int evdev_root_x;
int evdev_root_y;
int evdev_button;

int evdev_key_val;

/**********************
 *      MACROS
 **********************/

/**********************
 *   GLOBAL FUNCTIONS
 **********************/

/**
 * Initialize the evdev interface
 */
void evdev_init(void)
{
	char str[2048];
	int input_device = 0;
	char *ptr;
	
	// search for touch screen
	// first check which device need to be opened.
	do
[... 181 lines cut ...]
    data->point.y = evdev_root_y;
#endif

    data->state = evdev_button;

    if(data->point.x < 0)
      data->point.x = 0;
    if(data->point.y < 0)
      data->point.y = 0;
    if(data->point.x >= drv->disp->driver->hor_res)
      data->point.x = drv->disp->driver->hor_res - 1;
    if(data->point.y >= drv->disp->driver->ver_res)
      data->point.y = drv->disp->driver->ver_res - 1;
	return ;
}

/**********************
 *   STATIC FUNCTIONS
 **********************/
int map(int x, int in_min, int in_max, int out_min, int out_max)
{
  return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
}

#endif

RichNeese, 2023-08-29:
Ok touch works but we have to get it working correctly. As the touchpoint are off

RichNeese, 2023-08-29:
in the install_guide.txt last line is wrong

//Run sdrberry
./sdrberry

should it not be sdrberry.sh ?

RichNeese, 2023-08-29:
I am going to start a fesh instll this am also.

RichNeese, 2023-08-29:
DO you plan to add bluetooth support for bluetooth headsets .  do you plan to add vox ?  I will be reading code  but I dont see what is and is not supported

paulh002, 2023-08-29:
it should be sdrberry without ./

paulh002, 2023-08-29:
I have supported bluetooth but it creates a lot of delay. So if you change
the frequency with an optical encoder the sound is delayed and makes it
unusable.
That is the reason I removed it. You can add bluetooth sound using
pulseaudio, but that is a bit of a challenge. But it does work.

paulh002, 2023-08-29:
Vox is a good idea, I will look into it. I use a icom sm30 microphone and
switch to transmit using a cat command.
I use a ESP32 board which connects to an optical encoder and also has rx/tx
switching. It would also be possible to use a GPIO pin.
An alternative is a video editor wheel like Contour ShuttleXpress they are
not so expensive. I am also checking some cheap usb encoder from aliexpress
https://nl.aliexpress.com/item/1005005226135891.html?spm=a2g0o.store_pc_allProduct.8148356.1.3e5c6f9dK1tuyt&pdp_npi=3%40dis%21EUR%21%E2%82%AC%2028%2C84%21%E2%82%AC%2014%2C42%21%21%21%21%21%40211b812f16933085829831116e250c%2112000032265852203%21sh%21NL%21173531318

RichNeese, 2023-08-29:
# Uncomment the WaveShare screen you use.

#Enable Waveshare 7inch_DSI_LCD_(B) 800x460
#cat >> EOF << /boot/config.txt
#dtoverlay=vc4-kms-v3d
#dtoverlay=vc4-kms-dsi-7inch
#EOF

#Enable Waveshare 7inch DSI LCD (C) 1024×600 driver：
#git clone https://github.com/waveshare/Waveshare-DSI-LCD
#cd Waveshare-DSI-LCD/6.1.21/64
#sudo bash ./WS_xinchDSI_MAIN.sh 70C I2C1

Adding this to test that would make it easier to add the touchscreens

paulh002, 2023-08-29:
I don't understand your remark. But I think it is about the rotation of the
touchscreen.
Best to disable the kms driver and use the legacy video driver. See
https://www.raspberrypi.com/documentation/accessories/display.html
The legacy driver is a bit easier to manipulate the touch screen.
Make sure the touchscreen and mouse have the same rotation.

RichNeese, 2023-08-29:
New Debian images  for pi 
https://raspi.debian.net/tested-images/￼

These images are for bookworm. 

I am testing on bookworm now . Also I have been making you install scripts more stable .

1) moving builds to /usr/src where they should be done.

2) addend checking that when it cd to a dir if it does not exist it exits the script and stopes building/installing.

. I will post once tested.

closed 2023-08-30 by paulh002