paulh002/sdrberry #28: Support for rpitx

hiperiondev, 2023-09-02:
Do you have any plans to support RPITX?
https://github.com/F5OEO/librpitx

paulh002, 2023-09-02:
Sdrberry supports (almost) every SDR that has an SoapySDR driver. I don't know this project, Is it like transmitting using the GPIO directly? That would be interesting. The most easy way to support is to have a SoapySDR driver for librpitx. Building a SoapySDR driver is relative simple, the librpitx project should implement this. In this way the SDR can be used by multiple SDR software projects

hiperiondev, 2023-09-02:
There is no SoapySDR support for RPITX that I know of.
Whether there is support for  GNU Radio: https://github.com/jmfriedt/gr-rpitx
An alternative would be to make this support. A long time ago I made an implementation for OpenHPSDR, it works correctly but it is a bit unstable to frequency change: https://github.com/hiperiondev/hpsdr-p1-rpitx

paulh002, 2023-09-02:
The documentation of the library is very minimum, SoapySDR requires API for setting gain (optional), frequency and sending the samples. That is all. You can check my library for audio / tayloe QSD/QSE SoapyHifiBerry. If these API's are defined it should not take much to create a SoapySDR driver.

hiperiondev, 2023-09-06:
That is probably the best option.
For a while I had a conversation about the documentation with the author but he never added anything about how it works.
To better understand how it works and correct some problems I made a C translation of the library.
When I have a bit of time I will surely try to create support for SoapySDR

paulh002, 2023-09-06:
If you have a first version i am happy to test it.

hiperiondev, 2023-09-08:
I will reopen when I have something working

hiperiondev, 2023-09-11:
I have written the skeleton of the driver based on the PlutoSDR driver
Responds correctly to detection.
When I have something functional I'll let you know.

https://github.com/hiperiondev/SoapySDR_rpitx

paulh002, 2023-09-12:
The PlutoSDR driver is a good starting point, I also used it. I have to do some research on rpitx how they achieve to create that high frequency. Maybe we can also use it as a LO source for a tayloe mixer. That would be interesting.

hiperiondev, 2023-09-16:
It already works... but with certain transmission outage problems.
I think it is because of the buffer loading thread.
But you can now try:
https://github.com/hiperiondev/SoapySDR_rpitx

paulh002, 2023-09-17:
I have created a fork of the project to test it. I am not 100% sure what is the best solution, because the SoapySDR driver for rpitx does not have a RX channel. In principle it should report 0 channels for RX. But Sdrberry can not handle that currently, this I am fixing now first. But in the end I think it would be nice to support 2 SoapySDR drivers at the same time. One for TX and one for RX. But that will take some time. Once I have things working I will update the repository.

hiperiondev, 2023-09-17:
I have allowed the Soapy driver to report the RX channel because I have encountered several programs that require it. It does not have functionality but it exists to avoid this problem.

paulh002, 2023-09-17:
In that case you should also support the streaming of rx data, in the right
timing fashion. For sdrberry I prefer that the SoapySDR driver reports no
rx channels.
Maybe add a configuration option or a compile define option to select the
behaviour.
Maybe you can also add cmake support, the installation of the driver
depends on the version of raspberry pi OS,
I will also try to mae a cmake file, because it makes cross compilation
easier.

paulh002, 2023-09-17:
I get some output, but the frequency does not match. I will wait for your update.

[image]

hiperiondev, 2023-09-21:
I have made a lot of progress and it already transmits fairly well. I have found that the rate of consumption of the transmission FIFO in RPITX is not constant.
I hope to be able to solve it within a short time.

chrisdebian, 2025-03-16:
Does this issue need to remain open?

closed 2025-03-16 by paulh002