paulh002/sdrberry #31: TX won't work on FM mode with hackrf SDR

jacintomfr, 2024-05-03:
Hi, 
I decide to try your project and the script worked at first time, I'm using a hackrf to test and the RX and TX works OK on mode SSB but TX don't work well in FM mode because when I switch off TX the program freeze and I have to turn off and on again the pi.
Your project is really interesting, would like to try more, any chance to have kicad information about the esp32 controller ? 
tks

Jacinto
CU2ED

paulh002, 2024-05-03:
Hello Jacinto,

Thanks for your interest. I will check the FM transmission, i can replicate the issue so I will try to fix it.

paulh002, 2024-05-03:
I fixed the issue, you update sdrberry by using git pull and compile the application in the build directory. Use sudo make install to install the new version. (Or reexecute the install script.)

jacintomfr, 2024-05-04:
Hi, 

tks for check and for the ep32 kicad files ... i did a git pull and compile and indeed it fixed but when I try to speak at microphone there is modulation but can't understand it seems like sample rate is not right, as far as I could read for hackrf I have to choice for rx and tx 4Ms ? right ? I selected this on setup menu and on tx menu ... btw how about  a squelch option for FM mode, that will be great  :-)
For external controller, midi is not a bad choice, more easy but I also like the idea to have SWR meter :-)
 
Jacinto
cu2ed

jacintomfr, 2024-05-04:
Hi, I think I messed up with the samples rates so I decided to copy from the file sdrberry_settings_hackrf.cfg (like a reset) and now it works FM modulation is good but it seems the RX the audio is very low  ... I tried with my fixed and mobile radio ... when possible can you confirm ?
I also detected that as soon I change span from the default 2000khz for lower the TX still works but RX there is problems I can't receive then a couple of seconds later I receive the signal, only on 2000khz RX works well but audio low ...

Jacinto
cu2ed

jacintomfr, 2024-05-04:
Hi, 

after a few more tests it seems if the span < 96khz the audio is OK but the latency is higher on audio and the spectrum and waterfall it's not on the right frequency i mean it's not syncronized ... can you reproduce this ? if not it's my mistake but I can't figure out where is my problem ...

tks

Jacinto

paulh002, 2024-05-04:
Hi Jacinto,

I did not do so much work on FM modulation, also the hack rf is a bit of a problematic device (for me). I don't know why but compared to other devices like pluto adalm it has more issues. I have to setup a test environment with the hack rf (I have one) to dive more into the issues. The delay is caused because of variation in the samplerate, in SSB I compensate for this problem. I will copy that code also to FM. Span should not have any influence on the audio, it is only for display. The only thing I have to test is the optimum samplerate for the HackRF 4Mbit or 2Mbit is a bit high for the raspberry pi. Something like 500Khz would be better, but as I remember the HackRF had problems with lower sample rates.

paulh002, 2024-05-04:
I have updated the samplerate compensation in the build version, you can try it by doing a git pull and a git checkout build. Compile this version and see if the delay is not happening any more.

jacintomfr, 2024-05-04:
Hi, 
that will be great, I'm really interested in this project and hope you continue to develope.
I do have 1 pluto and also 2 radioberry but pluto is dedicated to DATV and radioberry is only HF and I use pihpsdr that also can use soapy driver but it don't work very well with hackrf, and because I have 2 hackrf this seems to me the best solution to use this device, more in VHF and UHF bands ... as I said FM modulation on TX is very good and very low latency :-)
About hackrf sample rate, yes it's true on my projects with DATV on gnuradio I had to use >3Ms so I can get best performance.
>> Span should not have any influence on the audio, it is only for display
yes, like a zoom on the display, right ? but at the moment I have huge inlfuence on audio at last with me in SSB and FM.
Now I will check esp32 controller ... I do have some parts ... but I need to buy the esp32.

OK, it seems we have an update, will try and I give you later a report.

tks

Jacinto

paulh002, 2024-05-04:
Yes, I am still developing this project. Right now I am mostly focusing on a Tayloe HF board (see SoapyHifiBerry). But I also have radioberry and SDRPlay etc. I still have PCB's for ESP32 if you like I can send one. I also will look into how to implement a squelch should not be so difficult. 

73

jacintomfr, 2024-05-04:
Hi, 

I'ma happy that you want to continue, we can have cheap solution for 6m to 13cm with hackrf or pluto (if it can work well at
network conection insted USB, and also with the last firmware mod, I can also make some tests) low power (+- 10w)

I'm not very familiar with github, so I see you change to build branche
I have done:

cd sdrberry/build
git pull
git checkout build
make -j4
sudo make install

no erros but FM (RX) is horse I mean when I'm in TX I can hear an high pinch audio on speaker (no mute) and the problem with span is still present, it seem only with everything at 2000Ms it works OK lantency not bad and waterfall + spectum sync, still audio low ... there is DC at center but this is normal it seems it can be removed at least sdrangel can do that ...
noise floor and speed 1 don't do anything but when you have the hackrf test setup OK you can verify that.

About the PCB that will be great, you have my info at qrz.com send me later a e-mail so we can arrange more details ... tks

Jacinto
cu2ed

paulh002, 2024-05-04:
almost ..
To update execute:

cd sdrberry
git pull
git checkout build
cd build
make -j4
sudo make install

paulh002, 2024-12-31:
Hi, I did some fixes for hackrf. The main issue was that the hackrf is halfduplex. Also the the span issue should be ok now. The resolution and rotation of display is managed now in the sdrberry_config.cfg. (See the wiki page). Also make sure that the option halfduplex=1 is set in [hackrf] group.

73,
Paul

jacintomfr, 2024-12-31:
Hi Paul, tks for the hackrf updates, I will definitely check this ...

73

Jacinto

closed 2025-03-05 by paulh002