paulh002/sdrberry #32: Can install driver for radioberry

jacintomfr, 2024-05-04:
Hi, 

I was trying to install radioberry driver but I get this error:
make[2]: *** [scripts/Makefile.build:250: /home/pi/Radioberry-2.x/SBC/rpi-4/device_driver/driver/radioberry.o] Error 1
make[1]: *** [Makefile:2012: /home/pi/Radioberry-2.x/SBC/rpi-4/device_driver/driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
make: *** [Makefile:8: radioberry.ko] Error 2
cp: cannot stat 'radioberry.ko': No such file or directory
modprobe: FATAL: Module radioberry not found in directory /lib/modules/6.1.21-v8+
chmod: cannot access '/dev/radioberry': No such file or directory
modinfo: ERROR: Module radioberry not found.

even if I try Joahn driver I also get this:
make[2]: *** [scripts/Makefile.build:250: /home/pi/Radioberry-2.x/SBC/rpi-4/releases/dev/Radioberry-2.x/SBC/rpi-4/device_driver/drive r/radioberry.o] Error 1
make[1]: *** [Makefile:2012: /home/pi/Radioberry-2.x/SBC/rpi-4/releases/dev/Radioberry-2.x/SBC/rpi-4/device_driver/driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
make: *** [Makefile:8: radioberry.ko] Error 2
Radioberry driver installation failed.

You are using a linux version without linux header files; choose an other distro.

What os version is right for radioberry ? I'm confuse
I do have interface juice (USB) for radioberry and the process is diferent

Jacinto
cu2ed

paulh002, 2024-05-04:
Make sure you use 64 bit bullseye, not 32 bit. Best to use my installation
script for radioberry.

jacintomfr, 2024-05-04:
Well, this is not easy ...

uname -a
Linux sdrberry 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

I'm using your https://github.com/paulh002/Radioberry-2.x
and I get this error:
/tmp/Radioberry-2.x/SBC/rpi-4/device_driver/driver/radioberry.c:361:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  361 |  int result = platform_driver_register(&radioberry_driver);
      |  ^~~
make[3]: *** [scripts/Makefile.build:250: /tmp/Radioberry-2.x/SBC/rpi-4/device_driver/driver/radioberry.o] Error 1
make[2]: *** [Makefile:2012: /tmp/Radioberry-2.x/SBC/rpi-4/device_driver/driver] Error 2
make[1]: *** [Makefile:8: radioberry.ko] Error 2
make: *** [Makefile:26: SBC/rpi-4/device_driver/driver/radioberry.ko] Error 2

what I missing ?
I'm using OS image : 2023-05-03-raspios-bullseye-arm64-lite.img

Jacinto

paulh002, 2024-05-04:
Sorry, you should use bookworm 64 bits.
Just use the sdrberry instal script for radioberry
Or go to /Radioberry-2.x/SBC/rpi-4/device_driver/driver
do make, and then sudo make install

Make sure the radioberry service is not running (this simulates hermes
light 2 interface)
you can check with sudo systemctl status radioberry
just disable this firmware driver sudo systemctl stop radioberry
sudo systemctl disable radioberry

jacintomfr, 2024-05-04:
OK, but I think I was using wrong OS version, after reading your files:
The software is tested on Raspberry PI Model 4B and the 64 bit version of rasperry pi OS with the GUI switched of.
                                                                                                                                                             
I was using lite version ... maybe the problem ...
anyway I will switch to bookworm 64 bits full img with GUI off and see the result ...

tks again

Jacinto

paulh002, 2024-05-04:
Yes, just install 64 bit bookworm with gui, then use raspi-config to switch
it off.
Also before installation apply all updates.

jacintomfr, 2024-05-04:
Hi, 

now I think everything is instaled as you can check (bottom) but I get error when try to load sdrberry:

line 4: 1034 aborted ...

will giveup for today ...

Jacinto

Installing Radioberry gateware Cyclone 10 CL025...

Radioberry gateware Cyclone 10 CL025 installed.
Installing Radioberry driver...
make[1]: Entering directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
  CC [M]  /home/pi/Radioberry-2.x/SBC/rpi-4/device_driver/driver/radioberry.o
  MODPOST /home/pi/Radioberry-2.x/SBC/rpi-4/device_driver/driver/Module.symvers
  CC [M]  /home/pi/Radioberry-2.x/SBC/rpi-4/device_driver/driver/radioberry.mod.o
  LD [M]  /home/pi/Radioberry-2.x/SBC/rpi-4/device_driver/driver/radioberry.ko
make[1]: Leaving directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
filename:       /lib/modules/6.6.20+rpt-rpi-v8/kernel/drivers/sdr/radioberry.ko
version:        95
license:        GPL
description:    Radioberry SDR device driver. (rpi-4)
author:         Johan Maas - pa3gsb@gmail.com
srcversion:     22A4BA62C709AEAE85D9445
alias:          of:N*T*Csdr,radioberryC*
alias:          of:N*T*Csdr,radioberry
depends:
name:           radioberry
vermagic:       6.6.20+rpt-rpi-v8 SMP preempt mod_unload modversions aarch64

Radioberry driver installed.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'pulseaudio' is not installed, so not removed
The following package was automatically installed and is no longer required:
  libraspberrypi0
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--2024-05-04 21:10:41--  https://raw.githubusercontent.com/paulh002/sdrberry/master/install/sdrberry_settings_radioberry.cfg
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2662 (2.6K) [text/plain]
Saving to: ‘sdrberry_settings_radioberry.cfg’

sdrberry_settings_radioberry.cfg                100%[====================================================================================================>]   2.60K  --.-KB/s    in 0s

2024-05-04 21:10:42 (7.32 MB/s) - ‘sdrberry_settings_radioberry.cfg’ saved [2662/2662]

paulh002, 2024-05-04:
Reboot and start sdrberry with sudo sdrberry. Radioberry driver need root
rights to start. If it does not work share me the startup log

jacintomfr, 2024-05-04:
Hi, it's working ... 

with SoapySDRUtil --find it found radioberry but when I do --probe... I get error 
after a few checks I detected that I forgot to turn on SPI protocol :-(

sorry for that

now everything is OK

Jacinto

paulh002, 2024-05-04:
Great

closed 2024-05-04 by paulh002