paulh002/sdrberry #36: sdrberry

Marc1950, 2024-11-28:
Hello 
I installed your application but it does not launch in console mode like in your video. 
73 Marc F6GKG

paulh002, 2024-11-28:
I need some more information, what sdr are you using? Can you share the log
output? What type of Pi? Operating system etc. Then I can help better

Marc1950, 2024-11-29:
Hello Paul
Thanks for your answer. The SDR used is the RadioBerry with a Raspberry 4 and the OS is the Raspberry PI 64 bits.
I must tell you that I know nothing about programming Raspberry. For the Log file tell me where I should find it.
(Google translation).
73

paulh002, 2024-11-29:
For radioberry please first check the discussion: No display or audio #35
https://github.com/paulh002/sdrberry/discussions/35

sdrbery probebly cannot access the radioberry due to firmware that is
started or you need to start it with sudo

73
Paul

Marc1950, 2024-11-30:
Hello Paul
I formatted the SD card and started a clean installation of raspberry and SDRBerry in SSH with Putty.
SDRberry starts provided that I add the line on the console:
**sudo sdrberry > sdrberry.log 2>&1**
And not automatically
And on the banner at the top: No SDR device found
[image]
73

paulh002, 2024-11-30:
Somehow the radioberry driver is not working correctly. You can check the driver from the command line with the command ```dmesg | grep radioberry```    Please check if the gateware is loaded correctly. You can also test SoapySDR with the command ```SoapySDRUtil --probe``` Can you also share the sdrberry.log file?

Marc1950, 2024-11-30:
The command "dmesg | grep radioberry" doesn't tell me anything. 
For the SoapySDRUtil --probe I attach the file.
For the gateware I don't know how to do it.
For the log file tell me where I can look for it.
(Google translation)
[SoapySDR .txt](https://github.com/user-attachments/files/17964981/SoapySDR.txt)

Marc1950, 2024-11-30:
I found it
[sdrberry.log](https://github.com/user-attachments/files/17964999/sdrberry.log)

paulh002, 2024-11-30:
Hi, you installed the wrong SDR driver and SoaySDR driver. I think you used the wrong option when running the installation script. To install the radioberry driver use ```./install.sh RDB DSI```

Marc1950, 2024-11-30:
Indeed that was it, it works now but no sound comes out of the speakers.
And it still does not launch automatically in the console.
Thanks for your patience.

paulh002, 2024-11-30:
For sound you need an external USB Audio adapter. The build in audio does not have audio input. In the setup page there is an audio dropdown which shows the possible audio devices. If the device does not work. You should test it first from the commandline with aplay. Use ```aplay -l ``` to list all devices. You can test the sound device with ```ffplay -ar 48000 -f lavfi -i "sine=frequency=1000" -nodisp''' This should output a 1khz tone.

paulh002, 2024-11-30:
For autostart you have to use crontab. In the installation directory there are 2 files crontab and startsdr.sh copy both to the home directory. Make startsdr.sh executable ```chmod +x sdrstart.sh``` and load crontab ```crontab crontab``` The it should autostart after reboot

Marc1950, 2024-12-01:
Hello Paul 
For the automatic start it's good, it works.
On the other hand I would like to enlarge the screen but I can't find the solution
despite the instructions on github
73

paulh002, 2024-12-01:
You can change the resolution in sdrberry.cpp
Change the constant value's to the correct size.
```
const int screenWidth = 800;
const int screenHeight = 480;
```

Recompile and reinstall the application using make
```
cd build
cmake ..
make -j4
sudo make install
```

Marc1950, 2024-12-02:
Hello Paul
I'm sorry to bother you again.
I have changed the screen resolution in sdrberry.cpp and saved but for the following commands I get this:

Last login: Mon Dec  2 08:03:13 2024 from 192.168.1.87
marc@raspberrypi:~ $ cd build
cmake ..
make -j4
sudo make install
CMake Warning:
  Ignoring extra path from command line:

   ".."

CMake Error: The source directory "/home/marc" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** Pas de cible spécifiée et aucun makefile n'a été trouvé. Arrêt.
make: *** Aucune règle pour fabriquer la cible « install ». Arrêt.
marc@raspberrypi:~/build $

paulh002, 2024-12-02:
You have to be in the sdrberry directory.
cd sdrberry
cd build
cmake ..
make -j4
sudo make install

Marc1950, 2024-12-05:
Hello Paul
Just to let you know that everything is working fine so far. I just have to take care of the BF part.
I will let you know.
It was hard for me, thanks again for your patience.
73 by Marc F6GKG

paulh002, 2024-12-05:
Thanks I appreciate your feedback. Indeed it takes some effort to get it working.  Let me know if you have some ideas for improvement or interesting requirements.  73 Paul PA0PHH

Marc1950, 2024-12-06:
Hello Paul
Following on from the USB audio adapter, do you need a BF amplifier???.
73

paulh002, 2024-12-06:
What is a BF amplifier?

Marc1950, 2024-12-06:
Translation problem I meant Audio amplifier. But ignore my question because I plugged in a USB headset and it works.
However my screen is touch and it doesn't work.
And another question can we exit the application to return to the console.

paulh002, 2024-12-06:
It does not have a stop function. But you can kill the process with ```sudo
pkill sdrberry```
The touch screen should work with the build version of sdrberry. I solved
this issue for another user but did not merge it into the main branch yet.
To use the build version you have to checkout the build branch and
recompile.
```
cd sdrberry
git checkout build
cmake ..
make -j4
sudo make install
```

Not to forget to change the width and height for your display.

73
Paul

Marc1950, 2024-12-06:
I'm sorry about the touch screen can you give me more details I don't understand.
Again I'm sorry to bother you
73 Marc

paulh002, 2024-12-06:
No problem. Somebody else had the same problem. That i fixed in a different version of sdrberry. The "build" version. In this version I test new functions. You can switch to this version using git checkout. This will copy the files of the build version.

```
cd sdrberry 
git checkout build
cd build
cmake ..
make -j4
sudo make install
```
Then you have de version that should work with the touch interface

Marc1950, 2024-12-07:
Hello Paul
It doesn't work the touch screen doesn't work

paulh002, 2024-12-07:
Please share the log

Marc1950, 2024-12-07:
here is
[sdrberry.log](https://github.com/user-attachments/files/18048020/sdrberry.log)

paulh002, 2024-12-07:
I added the touch screen you use. You have to get the changes and recompile.
```
cd sdrberry
git pull
cd build
cmake ..
make -j4
sudo make install
```

Let me know if it works

Marc1950, 2024-12-07:
No, it doesn't work.
I just run the command lines you gave me with Putty.
Is there anything else I need to do???

paulh002, 2024-12-07:
Can you try again?
```
cd sdrberry
git pull
git checkout build
cd build
cmake ..
make -j4
sudo make install
```

and share log?

Marc1950, 2024-12-07:
still the same, here is the log
[sdrberry.log](https://github.com/user-attachments/files/18048627/sdrberry.log)

paulh002, 2024-12-07:
Bit experimenting, can you try again same as last time?

paulh002, 2024-12-07:
If you do a git pull you should see something like this:
[image]
Maybe not all but atleast the file evdev.c should be updated.
It should look like this 
[image]

Marc1950, 2024-12-07:
I get this "up to date"
[image]

paulh002, 2024-12-07:
That looks ok, can you also show 
```
git status
```

Marc1950, 2024-12-07:
[image]

closed 2024-12-07 by Marc1950

paulh002, 2024-12-07:
Ah, the problem is git.
Please try:
```
cd sdrberry
git stash
git pull
git checkout build
cd build
cmake ..
make -j4
sudo make
```

You should have to change sdrberry.cpp manual again

Marc1950, 2024-12-07:
sorry Paul it's still the same and the screen hasn't changed size

paulh002, 2024-12-07:
Then you still have the old code. Then delete the whole sdrberry directory
and clone it again.
rm -rf sdrberry
git clone https://github.com/paulh002/sdrberry
cd sdrberry
git checkout build
mkdir build
cd build
cmake ..
make -j4
sudo make install

paulh002, 2024-12-07:
> I followed your instructions the screen size has changed.
> but the touch works very poorly and very randomly.

Probably rotation of the touch interface. Sometimes you can rotate it in
the driver configuration. But I don't have that information. But you can
also change the code in sdrberry to rotate. I can send an example tomorrow.

Marc1950, 2024-12-07:
Ok thanks again for everything sorry to take up your time. I have to stop for tonight. Maybe tomorrow.
73 by Marc F6GKG

paulh002, 2024-12-08:
To swap touch xy axes you can change lv_drv_conf.h file (line 402) and set #define EVDEV_SWAP_AXES 1
Maybe this helps

Marc1950, 2024-12-08:
ok thanks Paul I will try tomorrow.

paulh002, 2024-12-09:
Then it is xy swap of the touch screen. Maybe you can check if the buttons are upside down if you add set #define EVDEV_SWAP_AXES 1. In that case you could add at line 254 (end of the function) ```data->point.y = 480 - data->point.y```
Replace 480 with the height of your display. You can do the same with x ofcause if necessary.
An other option is to change the config.txt and add  rotation and switch of the kms video driver  
```
#dtoverlay=vc4-kms-v3d
display_lcd_rotate=2
```

Marc1950, 2024-12-09:
here is the line of the modified file you will say if it is correct. For the second option I did not find the config.txt file
[image]

Marc1950, 2024-12-09:
I don't know if you could see the line, I wrote at the end.
 It's always the same, the touchscreen doesn't work well.

paulh002, 2024-12-09:
Your in the wrong file. The change has to made in lv_driver/evdev.c

Marc1950, 2024-12-09:
I made the change but it doesn't work.
[image]

paulh002, 2024-12-09:
First check how the touch screen is rotated.  Then try to set #define EVDEV_SWAP_AXES 1 in lv_drv_conf.h and compile and check if there is any change in touch location. If necessary you can manual mirror x or y or both if necesary. 

[image]

paulh002, 2024-12-09:
?

Marc1950, 2024-12-09:
Good evening Paul
Ok for all that. Yes I think it will be the best solution.
Moreover as I am ignorant in programming it is not made to help you. Do not hesitate to contact me again as soon as you have the solution. In the meantime I will use the mouse.
Happy holidays and see you soon.
73 by marc F6GKG

paulh002, 2024-12-14:
Hi Marc,
You can try again. Setting the resolution is now managed in the configuration file as is the xy orientation of the touch screen.
Edit the file sdrberry_config.cfg as described in the wiki
See https://github.com/paulh002/sdrberry/wiki/Input-devices
and
https://github.com/paulh002/sdrberry/wiki/Screen-resolution

To install new version:
```
rm -rf sdrberry
git clone https://github.com/paulh002/sdrberry
cd sdrberry
mkdir build
cd build
cmake ..
make -j4
sudo make install
```

Marc1950, 2024-12-14:
Hello Paul
Ok, received.
This weekend I am very very busy.
As I will have to be very focused, I will inform you as soon as I have done the tests.
Thanks again
73 Marc

paulh002, 2024-12-16:
> Hello Paul
> I wanted to do a clean install from scratch by reformatting the SD card.
> I reinstalled SDRBERRY from:
> wget
> https://raw.githubusercontent.com/paulh002/sdrberry/master/install/install.sh
> chmod +x install.sh
> ./install.sh RDB DSI
>
> and then:
> sudo sdrberry > sdrberry.log 2>&1
>
> But unfortunately the application does not start.
> The screen remains black. I did the manipulation three times and it's the
> same
> the screen is Black.
> Marc

Can you share the log?
If it does not start it is mostly radioberry that cannot be found.
I assume you tried a reboot.

paulh002, 2024-12-16:
Also before the installation.
Make sure to update the OS.

sudo apt update
sudo apt upgrade
sudo reboot
Then use raspberry-config to switch to cli auto login

Marc1950, 2024-12-16:
I did an OS update and reboots don't change anything.
I'm attaching the log
[sdrberry.log](https://github.com/user-attachments/files/18151783/sdrberry.log)

paulh002, 2024-12-16:
Your audio device is not working.
I just did a clean install and everything works directly.
Your audio device wants to use pulse audio but that is not installed when
using the install script.
Best to use a simple USB audio adapter.
Or add pulseaudio
sudo apt install pulsaudio

But I don't have experience with pulse audio, it is used for bluetooth
devices and gives a lot of delay.,

Marc1950, 2024-12-16:
I don't use a pulsaudio device, just a USB headset microphone. I don't understand the first time I didn't have this problem.

paulh002, 2024-12-16:
you can check the usb headphone with 
Use aplay -l  to list all devices. 
You can test the sound device with ffplay -ar 48000 -f lavfi -i "sine=frequency=1000" -nodisp 
This should output a 1khz tone.

Marc1950, 2024-12-16:
The USB audio output works very well, I have a 1Kz signal in the headset

paulh002, 2024-12-16:
Looks like the hdmi audio adapter gives a problem
Can you switch off hdmi sound?

sudo vi /boot/firmware/config.txt

and comment dtparam=audio=on
#dtparam=audio=on

[7 more replies]