paulh002/sdrberry #37: SDRberry

Marc1950, 2025-01-21:
Hello I installed SDR berry for a RadioBerry and I get as in the photo at startup.
My screen is a 7 inch resolution 1024x600.
73 by Marc F6gkg

[image]

paulh002, 2025-01-21:
Oh no, that number is not in the list :) I can add it. I will give you an update. Also check the wiki there i explain the resolution settings

Marc1950, 2025-01-21:
Good evening Paul
Yes thank you for adding it. 73 Marc

paulh002, 2025-01-21:
Do you have a link to the exact type of display? Or do you know the color
depth of the display ? I think it was 16 but a link to the seller would be
helpful

Marc1950, 2025-01-21:
ok I'll look at that tomorrow.
Have a nice evening

Marc1950, 2025-01-22:
Hello Paul
Here is some information about my screen.
https://52pi.com/collections/lcds-displays/products/7-inch-1024x600-60hz-ips-capacitive-touch-screen-with-speakers-for-raspberry-pi-windows-pc
The Wiki:
https://wiki.52pi.com/index.php?title=EP-0177
Mine has the ref EP-0178 which I could not find on their site.
It was purchased on Amazon.
Marc

paulh002, 2025-01-22:
Hi, I added the resolution. You can give it a try. Pull the update from git  and recompile with:
```
cd sdrberry
git pull
cd build
cmake ..
make -j4
sudo make install
```

Make sure to set the resolution in sdrbery_config.cfg to 3 (see the wiki for more details)

```
[screen]
resolution = "3"
```

73,
Paul

Marc1950, 2025-01-23:
Hello Paul
Ok received, just one question:
I don't have a sdrberry_config.cfg file but two sdrberry_settings.cfg one in my personal folder and the other in pi and they are pretty much identical.
In which one must be modified.
73 marc

paulh002, 2025-01-23:
You're right it is sdrberry_settings.cfg. You need to change the sdrberry_settings.cfg in the startup directory mostly that is /home/pi. If you have used sdrberry before better to copy the .cfg file from the install directory (based on the sdr you use). Or delete the file and edit after you have started sdrberry. It will create a new file.

Marc1950, 2025-01-23:
Ok thanks for the info. 
I use a Radioberry with SDRberry Touch2, later I will try a HackRF 
73

paulh002, 2025-01-23:
Ok, please check the wiki for hackrf details
https://github.com/paulh002/sdrberry/wiki/hackrf

paulh002, 2025-01-24:
That is ok, the settings file is loaded in the working directory of the user. So I think you made the correct change.
What is the issue with touch 2?  I don't understand

Marc1950, 2025-01-24:
see second picture
Something is wrong with the screen setting.

paulh002, 2025-01-24:
Is that a different display?

paulh002, 2025-01-24:
You can change the resolution to match the display resolution as described in the wiki

Marc1950, 2025-01-24:
No it's the same hardware.
I have two SD cards with each of the two versions

Marc1950, 2025-01-24:
Changing the resolution doesn't change anything.

paulh002, 2025-01-24:
Delete or rename all the sdrberry_settings.cfg files and start sdrberry. It will create a new settings file. Edit this file with 
[screen]
resolution = 2 

check the logfile it should show the resolution when sdrberry starts

sdrberry > log.txt

Marc1950, 2025-01-24:
OK yes you are right, thanks for the info, I know what Touch 2 means now. It doesn't matter I will use the other one.
I have to look now for the touch keys they do not work or very badly.
73 by Marc F6GKG

paulh002, 2025-01-24:
Check the wiki https://github.com/paulh002/sdrberry/wiki/Input-devices you can configure the touchscreen rotation

Marc1950, 2025-01-26:
Hello Paul
SDRberry works very well except the touch screen.
I tried all the settings in the file..cfg according to the Wiki nothing works. Except randomly on position 2.
On the wiki on the line touchscreen = "raspberrypi-ts"
And me on my file touchscreen = "ft5x06"
I did not try to modify.
73

paulh002, 2025-01-26:
For the ft5x06 you don't need to change anything.
It is probably the rotation of the coordinates.
First check the log file (by starting sdrberry > log.txt)
it should show something like:

Input device name: "pwr_button"
Input device name: "vc4-hdmi-0"
Input device name: "vc4-hdmi-0 HDMI Jack"
Input device name: "vc4-hdmi-1"
Input device name: "vc4-hdmi-1 HDMI Jack"
Input device name: "11-005d Goodix Capacitive TouchScreen"
Touch device name: "11-005d Goodix Capacitive TouchScreen"

If that is the case, than we have to figure out the rotation.
Maybe for test we can print the touch xy coordinates to check.

paulh002, 2025-01-26:
I added some print statments to the touch interface in the build branch.
To check the touch interface , you can do the following:

1 set touch_swap_xy = "0" in sdrberry_settings.cfg
2 build and install the build branch:

```
cd sdrberry
git pull
git switch build
cd build
cmake ..
make -j4
sudo make install
cd ~
sdrberry
```

Check the output of sdrberry log, and see what coordinate the left, top of the touch screen gives

Marc1950, 2025-01-26:
ok
i left my house.
i'll check later

Marc1950, 2025-01-26:
I followed your instructions but unfortunately I did not find any log file concerning sdrberry.
At the end of the command script the Rapsberry reboots.
And in the console I have this

[image]

:

paulh002, 2025-01-26:
The log file is the output of sdrberry you get a log file by redirecting
the output to a file. Start sdrberry with redirecton: sdrberry > log.txt

paulh002, 2025-01-27:
Ok, this is the correct file
The touch interface is selected

The framebuffer device was mapped to memory successfully.
Input device name: "ILITEK       ILITEK-TOUCH"
Touch device name: "ILITEK       ILITEK-TOUCH"

Did you check the touch interface is working and did you need to rotate the touch interface when you install Raspi OS?
So if you do a normal installation of the rapberry pi, and you get the standard desktop. Is everything working normal?

paulh002, 2025-01-27:
> No I don't need to rotate the screen when I install the Raspi OS.
> When I switch back to desktop mode the touchscreen also works fine.

What orientation does the gui have when you install Raspi OS?
landscape or portrait?

Marc1950, 2025-01-27:
Landscape

paulh002, 2025-01-27:
So in sdrberry_settings.cfg, you only set the resolution. No screen rotation?

paulh002, 2025-01-27:
Can you share the output of this command?

```
cat /proc/bus/input/devices
```

Marc1950, 2025-01-27:
In home/pi/sdrberry_settings.cfg it's already at 0
[screen]
resolution = 0
rotation = 0
And in home/personal folder/sdrberry_settings.cfg there is only:
[screen]
resolution = "2"

Marc1950, 2025-01-27:
[sdrberry.txt](https://github.com/user-attachments/files/18559673/sdrberry.txt)

paulh002, 2025-01-27:
You start sdrberry in home/personal folder/ ?
Because the settings file is red from the directory where sdrberry is
started.

paulh002, 2025-01-27:
Can we try again to use the build version?

```
cd sdrberry
git pull
git switch build
rm -rf build
mkdir build
cd build
cmake ..
make -j4
sudo make install
cd /home/personal folder/
sdrberry > log.txt
```

Then touch the screen, left upper corner.
And check / send the log.txt file

Marc1950, 2025-01-27:
> Other option is to check the touch screen with evtest
>
> sudo apt install -y evtest
> sudo evtest
>
> Select the touch screen number, and touch the screen
> Please share the output

ok
i have to go away
i'll try it this evening.

Marc1950, 2025-01-28:
Hello Paul
To answer your question I start SdrBerry in console mode
and the command "sudo /usr/local/bin/sdrberry >/dev/null 2>&1" when the problem is solved I will try the automatic mode.
You will find the log file attached.

[log.txt](https://github.com/user-attachments/files/18569200/log.txt)

paulh002, 2025-01-28:
It looks like x and yu coordinates of the touch interface are swapped. The y coordinate is clipping at 599. 
Can you set in sdrberry_settings.cfg the touch_swap_xy = 1 ?

```
[input]
touch_swap_xy = 1
```

and send the log.txt again?

Marc1950, 2025-01-28:
here is

[log.txt](https://github.com/user-attachments/files/18571114/log.txt)

paulh002, 2025-01-28:
The x and y coordinates are now within the correct range. Question now is if the zero point (0,0) is in the right corner. 
Can you touch only the left upper conner? And check the output or send it?

Marc1950, 2025-01-28:
here is

[log.txt](https://github.com/user-attachments/files/18571607/log.txt)

paulh002, 2025-01-28:
Ok, the x and y is also mirrored.
I added an extra translation to the touch interface.
Can you get the latest build version?
And set touch_swap = 4 

```
[input]
touch_swap_xy = 4
```

```
cd sdrberry
git pull
git switch build
rm -rf build
mkdir build
cd build
cmake ..
make -j4
sudo make install
cd /home/personal folder/
sdrberry > log.txt
``

Marc1950, 2025-01-28:
That's it, but the touchscreen still doesn't work.
The log file is attached

[log.txt](https://github.com/user-attachments/files/18574614/log.txt)

paulh002, 2025-01-28:
I think you did not update the version of sdrberry
this is the old behaviour. (only the new version has the 4 option for
swap_xy)

touch x: 404 y 599
touch x: 404 y 599
touch x: 404 y 599
touch x: 426 y 599
touch x: 448 y 599
touch x: 462 y 599
touch x: 462 y 599
touch x: 462 y 599
touch x: 462 y 599
touch x: 462 y 599
touch x: 462 y 599
touch x: 462 y 599
touch x: 462 y 599
touch x: 462 y 599
touch x: 473 y 599
touch x: 473 y 599
touch x: 480 y 599
touch x: 480 y 599
touch x: 476 y 599
touch x: 469 y 599
touch x: 466 y 599

Marc1950, 2025-01-28:
I forgot to indicate the path of the SdrBerry folder.
I resend the file.

[log.txt](https://github.com/user-attachments/files/18577078/log.txt)

paulh002, 2025-01-28:
Still the same, you did do a git pull? And rebuild sdrberry?

Marc1950, 2025-01-28:
I did

cd sdrberry
git pull
git switch build
rm -rf build
mkdir build
cd build
cmake ..
make -j4
sudo make install
cd /home/personal folder/
sdrberry > log.txt

paulh002, 2025-01-28:
Ok, I will check later why it is not ok

Marc1950, 2025-01-28:
Yes

[input]
mouse = "Mouse"
touch_swap_xy = "4"
touchscreen = "ft5x06"

Marc1950, 2025-01-28:
The build ends this way

[image]

And the SDRBERRY starts

Marc1950, 2025-01-28:
Here is the latest file
Something has changed

[log.txt](https://github.com/user-attachments/files/18578075/log.txt)

paulh002, 2025-01-28:
Ok, now it is updated but it almost looks like there are 2 sdrberries
running. If radioberrt can not be reached some other program is using it.
Make sure nothing is running. You can check with top command.

paulh002, 2025-01-28:
I updated sdrberry to check if an instance is already running. All changes are now available with normal installation. Best to reinstall. Or do a  pull and rebuild.

Marc1950, 2025-01-29:
I chose the reinstallation but unfortunately stopped at the beginning.

./install.sh: line 89: cd: liquid-dsp: No such file or folder
marc@raspberrypi:~ $

paulh002, 2025-01-29:
Please run the installation in /home/pi directory.
Do not use a different directory

Marc1950, 2025-01-29:
Here is Sdr Berry installed. Sorry for the inconvenience I had not given all the permissions to the pi directory.
I set input to 4 and screen to 2.
log file attached

[log.txt](https://github.com/user-attachments/files/18585730/log.txt)

paulh002, 2025-01-29:
There are still permission issues.
I keep debugging simple better keep everything standard.
So better not change installation directory, use different users etc.
To output the xy coordinates of the touch display you can set
[input]
touch_debug = "1"

See errors below:
correction rx 0, tx 0
80 m i2c  I2C time 0
tx gain 0 tx phase 0 rx gain 0 rx phase 0
Error writing "sdrberry_settings.cfg"
Error writing "sdrberry_settings.cfg"
Error writing "sdrberry_settings.cfg"
select_mode_rx stop all threads

closed 2025-03-16 by paulh002