paulh002/sdrberry #38: SDRPlay Segmentation Fault

TechMindsYT, 2025-02-26:
Hi there, when I run sdrberry with the sdrplay configured, I receive a "Segmentation Fault".

I first successfully had the HackRF working, looking nice on screen and receiving.

To try an RSPdx I performed the following:

1. Install the SDRPlay API for linux
2. Install SOAPYSDRPlay

Soapy SDRUtil can located the attached RSPdx connected via USB to the pi.

in sdrberry_settings.cfg I changed device to "sdrplay".

When I run "sudo sdrberry > sdrberry.log 2>&1" I now receive segmentation fault.

Is there another step to get SDRPlay devices working?

Thanks
Matt

paulh002, 2025-02-26:
It was some time ago I tested SDRPlay, I will test it this evening and let you know.

TechMindsYT, 2025-02-26:
Hi Paul, thank you for checking. I also notice the same issue with RTLSDR. The only module that works okay for me is hackrf.

paulh002, 2025-02-26:
That is correct, it is the same issue. The SDRPlay is now working fine with me. The check that SDRPlay and RTLSDR are receive only was not working any more. I fixed it now. But want to do some checks that I don''t break other SDR's before I release the fix. Thanks for testing.

TechMindsYT, 2025-02-26:
Brilliant! Please let me know once fixed and I will pull the update! Thanks for your hard work, I love the look of the UI

paulh002, 2025-02-28:
I am still testing but if you want to check you can use the build version.
If you have experience with git you can get the build version like this:
(If necessary use git stash, if there are local changes. This happens when using the new Pi Touch 2 display this display need 32 bit color) 

cd sdrberry
git pull
git switch build
cd build
cmake ..
make -j4
sudo make install

paulh002, 2025-02-28:
One remark on SDRPlay. There some issues I still have to check. One is the RF gain is inverted. That is large number is low gain. Normaly this should be managed in the SoapySDR driver but is not done. The 2nd thing is you have to set the bandwidth. Somehow the SDPlay driver is not accepting settings changes during startup. I have to discus this with SDRPlay API developers

paulh002, 2025-02-28:
> Hi Paul, Thank you for your work. I have tested your above instructions
> but once complete, and attempting to run for both SDRPlay and RTLSDR I
> receive "Aborted" on the command line and the application does not start.
>
> HackRF is still working okay though.
>
> Looking at the sdrberry.log file, the first debug complains something
> about a 0 samplerate. I can paste the entire log here if you like.
>
> Thanks,
> Matt

Yes please share the log and the cfg file also please

TechMindsYT, 2025-02-28:
`Option name: bass value: 4
Option name: fast value: 
Option name: medium value: 
Option name: mode value: 1
Option name: ratio value: 12
Option name: slow value: 
Option name: threshold value: 0
Option name: treble value: 0
ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused

Screen resolution 800 x 480 screenRotate 0 touch_swap setting 3
Input device name: "C-Media Electronics Inc.       USB PnP Sound Device"
Device is not a keyboard
Input device name: "RAPOO Rapoo 2.4G Wireless Device"
Device is not a keyboard
Input device name: "vc4-hdmi-0"
Device is not a keyboard
Input device name: "vc4-hdmi-0 HDMI Jack"
Device is not a keyboard
Input device name: "vc4-hdmi-1"
Device is not a keyboard
Input device name: "vc4-hdmi-1 HDMI Jack"
Device is not a keyboard
Input device name: "10-0038 generic ft5x06 (79)"
Device is not a keyboard
end keyboard 
Input device name: "C-Media Electronics Inc.       USB PnP Sound Device"
Input device name: "RAPOO Rapoo 2.4G Wireless Device"
Mouse device name: "RAPOO Rapoo 2.4G Wireless Device"

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi0,0), Unknown error 524.

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi1,0), Unknown error 524.

ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi0,0), Unknown error 524.

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi1,0), Unknown error 524.

129 device: Default ALSA Device input 0 output 10000
130 device: bcm2835 Headphones (bcm2835 Headphones) input 0 output 8
131 device: USB PnP Sound Device (USB Audio) input 1 output 2
No matching device found.

RtApi::openStream: input device ID is invalid.

ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused

Cannot open audio input stream

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi0,0), Unknown error 524.

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi1,0), Unknown error 524.

device name = Default ALSA Device
: maximum output channels = 10000
device name = bcm2835 Headphones (bcm2835 Headphones)
: maximum output channels = 8
device name = USB PnP Sound Device (USB Audio)
: maximum output channels = 2
No matching device found.

RtApi::getDeviceInfo: deviceId argument not found.

audio device = 0 USB Audio Device samplerate 48000 channels 0

RtApi::openStream: a non-NULL output StreamParameters structure cannot have an nChannels value less than one.

ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused
[... 208 lines cut ...]

[rtlsdr]
if-gain = "0"

[screen]
resolution = "0"
rotation = "0"

[sdrplay]
if-gain = 0

[web]
enabled = "0"

[wsjtx]
73 = ""
bandwidth = "4000"
call = "PA0PHH"
freqFT4 = {"1840", "3568", "5375", "7047", "10140", "14080", "18104", "21140", "28180"}
freqFT8 = {"1840", "3573", "5357", "7074", "10133", "14074", "18100", "21074", "28074", "28077"}
locator = "JO22"
reserverd = {"POTA"}
rx = "1200"
tx = "2300"
`

TechMindsYT, 2025-02-28:
I just managed to get it working by adding these lines in the .cfg file for the sdrplay stanza.

[sdrplay]
AGC = "off"
gain = "40"
if-gain = 22
samplerate = 2000

Is there a way to choose which Antenna port is in use as it appears to default to port A?

paulh002, 2025-02-28:
Ok, I will check that default values are added.
The antenna selection I can add, but I don't have a SDRPlay to test it.
So if you can give feedback, that would be great.

paulh002, 2025-03-01:
I added a separate SDR tab page. There is now a dropdown for the antenna. You can pull the latest build version.

TechMindsYT, 2025-03-01:
I Paul, I tried to compile the latest version and there appears to be an error with "undefined reference to guisdr".

Did I do something wrong? 

/home/pi/sdrberry/sdrberry.cpp:1139: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1139: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1139: undefined reference to `gui_sdr::get_decimation()'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1153: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1153: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1153: undefined reference to `gui_sdr::get_decimation()'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1117: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1117: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1117: undefined reference to `gui_sdr::get_decimation()'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1101: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1101: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1101: undefined reference to `gui_sdr::get_decimation()'
/usr/bin/ld: CMakeFiles/sdrberry.dir/sdrberry.cpp.o: in function `select_mode_tx(int, audioTone, int, int)':
/home/pi/sdrberry/sdrberry.cpp:1202: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1202: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1202: undefined reference to `gui_sdr::get_decimation()'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1215: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1215: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1215: undefined reference to `gui_sdr::get_decimation()'
/usr/bin/ld: CMakeFiles/sdrberry.dir/sdrberry.cpp.o: in function `gui_sdr::get_span()':
/home/pi/sdrberry/gui_sdr.h:39: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/gui_sdr.h:39: undefined reference to `guisdr'
/usr/bin/ld: CMakeFiles/sdrberry.dir/sdrberry.cpp.o: in function `switch_sdrreceiver(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/pi/sdrberry/sdrberry.cpp:1317: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1317: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1317: undefined reference to `gui_sdr::clear_sample_rate()'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1345: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1345: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1345: undefined reference to `gui_sdr::set_sample_rate(int)'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1372: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1372: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1372: undefined reference to `gui_sdr::init_bandwidth()'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1373: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1373: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1373: undefined reference to `gui_sdr::init_antenna()'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1317: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1317: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1317: undefined reference to `gui_sdr::clear_sample_rate()'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1335: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1335: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1335: undefined reference to `gui_sdr::add_sample_rate(int)'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1325: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1325: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:1325: undefined reference to `gui_sdr::add_sample_rate(int)'
/usr/bin/ld: CMakeFiles/sdrberry.dir/sdrberry.cpp.o: in function `main':
/home/pi/sdrberry/sdrberry.cpp:573: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:573: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:573: undefined reference to `gui_sdr::init(_lv_obj_t*, short, short)'
/usr/bin/ld: CMakeFiles/sdrberry.dir/sdrberry.cpp.o: in function `gui_sdr::get_span()':
/home/pi/sdrberry/gui_sdr.h:39: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/gui_sdr.h:39: undefined reference to `guisdr'
/usr/bin/ld: CMakeFiles/sdrberry.dir/sdrberry.cpp.o: in function `main':
/home/pi/sdrberry/sdrberry.cpp:678: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:678: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:678: undefined reference to `gui_sdr::set_radio(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:705: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:705: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:705: undefined reference to `gui_sdr::set_sample_rate(int)'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:734: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/sdrberry.cpp:734: undefined reference to `guisdr'
[... cut ...]
/usr/bin/ld: /home/pi/sdrberry/gui_tx.cpp:316: undefined reference to `guisdr'
/usr/bin/ld: /home/pi/sdrberry/gui_tx.cpp:316: undefined reference to `gui_sdr::get_current_tx_channel()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sdrberry.dir/build.make:1446: sdrberry] Error 1
make[2]: Leaving directory '/home/pi/sdrberry/build'
make[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/sdrberry.dir/all] Error 2
make[1]: Leaving directory '/home/pi/sdrberry/build'
make: *** [Makefile:139: all] Error 2

paulh002, 2025-03-01:
Forgot to update CMakeList.txt, I added the new file to CMakeList.txt. You can pull and retry again

TechMindsYT, 2025-03-01:
Ahhhh okay nice! I have it working! The only issue I have now is that I cannot seem to locate where to change the tuning STEP size? If I click on the frequency nothing pops up. Im sure it did with HackRF.

The Antenna selection appears to be working well too, but I will continue to test. Just a bit time consuming changing frequency without a direct dial or being able to change step size. Unless I have missed it?

paulh002, 2025-03-01:
Click the mouse wheel. You can also jump to the next or previous "page" so increase the frequency by the span size by clicking on the left or right edge of the frequency bar

paulh002, 2025-03-01:
sdrberry supports several HID devices like Contour ShuttleExpress or the cheap aliexpress volume knob. I use a separate ESP32 that sends CAT commands. But I could add support to connect an optical rotary controler to the GPIO directly.   Midi I don't use but could also be implemented.

TechMindsYT, 2025-03-02:
Hi Paul,

Thanks for your hard work and I can report back that it is working lovely with SDRPlay RSPdx. I will continue to test and play around with sdrberry, and I hope to make a video promoting your work very soon.

There are more features which SDRPlay support which would be really nice to have, although you mention you do not have a SDRPlay device. 

It would be nice to be able to (SDrPlay Devices Specific):

1. Enable or disable the onboard MW/FM and DAB filters
2. Control Bias Tee

Thanks again!

paulh002, 2025-03-02:
Hi Matthew,

Thanks for testing the program, some feedback is much appreciated. Until
now I have kept sdrberry low profile. The biggest challenge next to the
functionality is ease of installation and support for displays and touch
screens.
I just added support for the new pi Touch2, it is really a nice display.
Also it has a metal back plate which is good for RFI.
I do have a SDRPlay RSP1A which I use for testing.  I was already looking
at the other options for SDRPlay, there is an option for AGC it is in the
AGC tab.
But I will check how I can add these options, now with the extra tab for
the sdr device there is more space to add some options.

73
Paul PA0PHH

paulh002, 2025-03-03:
Hi,
I added a settings page in the sdr tab. It should display the items that you can enable and disable. You can just press the true / false and it will toggle the setting. You can pull the SDRPlay branch to check it.

cd sdrberry
git pull
git switch SDRPlay
cd build
cmake ..
make -j4
sudo make install

TechMindsYT, 2025-03-10:
Hi ya! I have had a chance to play around with this now and it works great. There are some instances where a crash happens, mainly when changing settings like bandwidth etc, but I will list them as they happen and report back.

When will SDRPlay and RTL support be merged into the master? I am making a video on this right at the moment and will start off by showing RTLSDR.

Cheers and thanks again for your hard work!
Matt

paulh002, 2025-03-11:
Hi Matt,
I am currently testing the latest patches for SDRPLAY and the installation script. I have added SDRPLAY as an option. When that is finished I will merge all to the main branch. The latest version is in the build branch.

paulh002, 2025-03-11:
I pushed the latest version to master branch

closed 2025-03-16 by paulh002