paulh002/sdrberry #39: Please add support for cariboulite

lohriialo, 2025-03-17:
Please add support for cariboulite

- https://github.com/cariboulabs/cariboulite
- https://www.crowdsupply.com/cariboulabs/cariboulite-rpi-hat

paulh002, 2025-03-17:
I did not test this SDR but looks very interesting for vhf and uhf use.
Cariboulite should work with the SoapySDR driver they supply.
You can add the sdr to the sdrberry_settings.cfg and try it.

[SDR Receivers]
default = "cariboulite "
receivers = {"hifiberry","radioberry", "plutosdr", "rtlsdr", "sdrplay", "hackrf", "cariboulite "}

[cariboulite ]
audiobuffersize = "4096"
bandwidth = "1750000"
dc = "0"
decimate = "0"
gain = "50"
if-gain = 21
samplerate = "1000"
samplerate_tx = "1000"
span = 192

heathhey, 2025-03-21:
Thanks for this information, I was going to request this as well. The one issue I am running into is that the cariboulite board parially blocks the DSI connector on my Pi4. Is there such a thing as a right angle ribbon connector?

paulh002, 2025-03-21:
Maybe there are, but more easy to obtain is a 20 pin extension connector for the pi, I use them for adding a passive cooling block to the pi. The pi can throttle when getting warm. 

[image]

closed 2025-03-26 by paulh002