paulh002/sdrberry #40: Crontab Not Working, and a few thoughts

scancapecod, 2025-03-19:
Hi Paul,

First off, many thanks for creating this. I saw the Tech Minds Youtube review and went to work from there. I'm happy to say I have it up and running with the SDRPlay RSP1B. 

I am having an issue when trying to auto start and following your instructions in the Wiki. When I run crontab crontab I am seeing the following error:

"crontab":30: command too long
errors in crontab file, can't install.

That is from the crontab downloaded from your installs directory. Not the end of the world but it would be nice to auto-start. Oh, and is there a "graceful" way to shut it down? If so I've missed that.

As to a few thoughts as your project develops, if possible:

1. In my case I would be using this for HF occasionally, but probably primarily as a truly portable receiver for viewing and monitoring VHF, UHF, and above signals, primarily narrow FM. Is there a way to add a squelch option so constant static is not heard when no signal is present?
2. Is there any chance to add a direct frequency entry keypad?

This is absolutely fantastic, and relatively easy to set up and use. Making some hit or miss edits to the sdrberry_settings.cfg file got me to the finish line and I've already learned how to change the band buttons, labels, and parameters to my liking.

Thanks very much and regards.

Scott/KC1UA
Cape Cod MA. USA

paulh002, 2025-03-19:
Hi Scott,

Thanks for your nice comments. The issue with the crontab file is probably because my text editor saved it in MSDOS format.  You can use the following command to fix it:

```
dos2unix crontab
```

Then try again.
In the sdrstart.sh there is a sleep function, you can edit the time. This is for allowing drivers to startup some SDR drivers need some time to start.

Thanks for the functional request good ideas. I have a keypad function for adding I2C commands so it should not be so much work to add a VFO keypad. Also the squelch is on top of my list. I was researching some algorithms for it. It looks a simple function but to make a usefull squelch seems to be a challenge.

Enjoy the software
73,
Paul/PA0PHH

scancapecod, 2025-03-19:
Hi Paul,

I tried as you suggest with the dos2unix command. It indicated that it changed it to Unix format, but when I tried it again I got the same error. To confirm, I'm running this in the order as shown in your Wiki:

crontab crontab
chmod +x sdrstart.sh
sudo reboot

I will admit that I know enough about Linux to get myself in trouble, so it could very well be that I'm doing something in error! It's not a big deal; I can start it manually, but as this will be the exclusive use of this setup having it autostart at some point will be ideal.

Thank you for your consideration of the feature additions. I'm sure the squelch issue is a difficult one as you likely have to take each SDR into consideration. Again, I'm not schooled enough in this stuff to know for sure, but your efforts are appreciated. I'm sure a lot of people offer, so I will do the same, if there is any testing I can offer with regards to the specific use of the SDRPlay devices I'd be happy to if I could.

Thanks again,
Scott

paulh002, 2025-03-19:
It should work I just tested it. Just for verification. The file sdrstart.sh should be located in the pi home directory. You are not using a different home directory or username? If so you have to edit the crontab file with the correct location.
The crontab file has this content, so sdrstart.sh should be in /home/pi location

```
@reboot sudo /home/pi/sdrstart.sh > /dev/null 2>&1
```

scancapecod, 2025-03-19:
Hi Paul,

The error is occurring when I run:

crontab crontab

This is the error message I get at that point:

"crontab":30: command too long
errors in crontab file, can't install.

This could be out of my overall ignorance of things.

Both the crontab and sdrstart.sh files are in my /home/pi directory. 

Thanks,
Scott

paulh002, 2025-03-19:
please check the crontab file it seems not to be correct. It should be like 

```
@reboot sudo /home/pi/sdrstart.sh > /dev/null 2>&1
```

scancapecod, 2025-03-19:
Interesting. I just had a look at it via nano and it has literally hundreds of lines worth of stuff in it. Maybe I should create a new crontab file with just that line in it?

paulh002, 2025-03-19:
> Everything in the crontab file is html code. I added the @reboot
> <https://github.com/reboot> line at the very end and then tried crontab
> crontab again from the command line, same error.
>
> Paul, this is not by any means critical and I don't want to take you away
> from other things, so there's no hurry on this; just not sure what is
> contained in that crontab file. Sorry for all of the confusion.

Delete the file and just make a new one with only that line

scancapecod, 2025-03-19:
We have a winner! That did the trick. Thanks for sticking with me!

paulh002, 2025-03-19:
No problem

paulh002, 2025-03-21:
Hi, I added a keypad you can try it.
You can get the update without installing again like this:

```
cd sdrberry
git pull
cd build
cmake ..
make -j4
sudo make install
'''
After this reboot

scancapecod, 2025-03-21:
Hi Paul,

I can report that the keyboard works great! However, it seems that the VFO button is not working now; can't switch from VFO1 to VFO2, button does not respond. 

Thanks!
Scott

paulh002, 2025-03-21:
VFO2 should work again. I will check the keyboard. I did not test it. But it should work if the keyboard is attached when the pi starts up. How ever I don't have extensive input checking when you use an external keyboard.

scancapecod, 2025-03-22:
VFO2 working fine now as stated, and both are tunable from the virtual keypad. No go with the external keyboard here but that's fine. Having direct frequency entry is great. Thanks!

On a separate note, is there a graceful shutdown of the app? If so I haven't found it and the only way I've been able to end it is to SSH in and do a shutdown.

Many thanks!
Scott

paulh002, 2025-03-22:
If you update sdrberry as before, the keyboard input should work.
You can type 14.1m [ENTER] to jump to 20m
[ESC] button will close the window.

scancapecod, 2025-03-22:
Hi Paul,

Confirmed keyboard is now able to input frequency on both VFO's. Thanks.

If you are saying the ESC key should allow me to exit sdrberry app and return to command line/terminal, that's not working. 

Also noticing something else that may or may not be related to these changes. Under the RX button, I am no longer able to scroll down to adjust the Spectrum Level. It will scroll low enough so I can see the top half of that wording, but the slider is not available.

I know that's starting to get off track of my original post; if you'd prefer I can post new "issue". I also have another question/observation regarding the RF and IF slider settings which I would discuss in a new thread, unless you want it here.

Thanks,
Scott

scancapecod, 2025-03-22:
Okay on RF/IF. As to scroll in RX, maybe you're using a different screen size than me? I'm using the original Raspberry Pi touch screen at 800 x 480. I definitely had access to the spectrum level before, not now for some reason.

paulh002, 2025-03-22:
Ok, the scroll it self works but you cannot see spectrum level?

scancapecod, 2025-03-22:
I can't access it because I can't see it. Screenshot attached.

[image]

paulh002, 2025-03-22:
Yes, I see. My current test system has the new Touch2 screen which has higher resolution. My tranceiver does have 800x480 but is in my shack so I missed it.

scancapecod, 2025-03-22:
I'll probably switch to the better screen at some point. Had I been thinking I'd likely have started with that; got keyed in on Matt's video that emphasized 800 x 480.

paulh002, 2025-03-22:
I updated the rx page

[image]

scancapecod, 2025-03-22:
Works great. Thanks!

paulh002, 2025-03-23:
Hi,
If you like you can try out the squelch, I added it only to FM (narrow band) for now.
You can get the version with squelch like this:

```
cd sdrberry
git pull
git checkout squelch
cd build
cmake ..
make -j4
sudo make install
```

scancapecod, 2025-03-23:
Hi Paul,

It seems to be working well. I need to do some testing on weaker signals. What is the significance of the bandwidth slider in relation to the squelch, or is that just for the AGC setting that's on that menu?

Thanks for doing this. Very significant for quietly monitoring FM traffic!

Scott

paulh002, 2025-03-23:
Hi Scott,

I still have to implement the bandwidth or delay for the AGC which is part of it, have to think what to do with the old AGC.
Maybe remove it  or swap it for an other function like EQ or so. For now I let it see how it works out.

73,
Pau;

scancapecod, 2025-03-23:
The squelch seems to be working quite well for me, to include being able to close it and still receive weaker signals. That's a balance of the IFGR and RFGR as well but I'm used to doing that with the SDRPlay devices regardless of the software being used. 

This is a really cool project. I had a few other thoughts on some enhancements but I don't want to come across as a pain in the backside either! I appreciate everything you've done thus far. Thank you!

Scott

paulh002, 2025-03-23:
If you have ideas they are welcome, can use some inspiration :). It also depends on the SDR you are using. I use mostly the radioberry which is an add on card for the pi. Which is a spinoff of the Hermes light 2. So TX capability is also important for me. 
I still need to do something with the PreAmp and Attenuator buttons, idea there is to connect using I2C to some attenuator or preamp boards. Sdrberry can already use I2C to select bandfilters using I2C. You can program it in the settings menu.
SoapySDR automatically distributes the gain over IFRG and RFGR. But you could also do this individual. Could add a slider to the sdr page, based in the gain names reported by the SoapySDR driver.
Right now my SDRPlay has problems on 40 meters with the very strong signals from broadcast stations. Maybe manual gain setting helps.

scancapecod, 2025-03-24:
I don't "think" what I'm thinking would be SDR dependent, but I'll post a new thread and we'll see if it's something feasible at some point. 

Thanks!

closed 2025-04-14 by paulh002