paulh002/sdrberry #42: Wierd display and no touch

hellberg78, 2025-03-21:
I am using a raspberry pi Touch Display 2
I have installed using HRF T2 
The display is in the right orientation but the SDRplay displayes wierd and the touch dosent work.

I think i have missed something but i cant figure out what!

Best Regards
Nicklas

[image]

paulh002, 2025-03-21:
Did you change the touch setting in cfg file? For T2 it should not be
changed. See the wiki page

hellberg78, 2025-03-21:
I started from the begining and now the touch is working BUT the display is stil messed up!
I have checked the resolution and that is set to 4

paulh002, 2025-03-21:
The touch 2 is 32 bit color, you should check in lv_conf.c if color depth
is set to 32. This should be automatically but maybe some issue

hellberg78, 2025-03-21:
I have checked this !

In lv_conf.c    : #define LV_COLOR_DEPTH 32

paulh002, 2025-03-21:
Also did you make any changes to cmdline.txt or config.txt? No changes should be made.

paulh002, 2025-03-21:
Can you also post the output of this command?

'''
fbset
```

hellberg78, 2025-03-22:
I have now done a new install with no changes to anything but still same problem.
Below you have the info you wanted!

[log.txt](https://github.com/user-attachments/files/19404440/log.txt)

[sdrberry_settings.txt](https://github.com/user-attachments/files/19404441/sdrberry_settings.txt)

output from fbset

mode "720x1280"
    geometry 720 1280 720 1280 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/0
endmode

paulh002, 2025-03-22:
Hi, 
This is a bit strange for a Touch2 display. It say it is a 16 bit display. In that case you have to use the DSI installation method, but set the resolution to 720x1280 like the touch2 display. 
What you can try is to change sdrberry to 16 bit color and try again.
Edit the file lv_conf.h and set colordepth to 16

```
/*====================
COLOR SETTINGS
====================/

/Color depth: 1 (1 byte per pixel), 8 (RGB332), 16 (RGB565), 32 (ARGB8888)/
#define LV_COLOR_DEPTH 32
```

Then recompile

```
cd sdrberry
rm -rf build
mkdir build
cd build
cmake ..
make -j4
sudo make install
```

See if this helps

hellberg78, 2025-03-22:
So that one did fix the problem!
Do you have any idea why it dossent work in 32bit if the display should support that?

[d89c_touch-display-2-product-brief.pdf](https://github.com/user-attachments/files/19404663/d89c_touch-display-2-product-brief.pdf)

paulh002, 2025-03-22:
I would almost say, return it. It is not good. I have the same display and
it shows 32 bit (it is 24 but will ignore first 8 bits)

paulh002, 2025-03-22:
I run it on a pi5, i can try a pi4 to see if it gives the same effect

akhepcat, 2025-03-25:
Do you have the following line defined in /boot/firmware/config.txt   : 

framebuffer_depth=16

??   that could be causing the issue