paulh002/sdrberry #44: SDRPlay - RF and IF Sliders

scancapecod, 2025-03-22:
Hi Paul,

This is more of a confirmation than anything else. Software that runs SDRPlay devices uses IF and RF settings to reduce gain, not increase it. I'm finding a wide variety of adjustments being made dependent on what frequency is in use. 

Just confirming when RF and IF sliders are moved to the right that the intention is indeed to reduce and not increase gain for either.

Thanks,
Scott

paulh002, 2025-03-22:
I think it is only the RF gain that is the wrong way if I am correct. The problem is, it is an issue with SDRPlay SoapySDR driver. The manufacture of SDRPlay has choosen to implement it like this, all other SDR's do it the other way. We should open the issue on SoapySDRPlay repo and ask them to correct it. So if I fix it for SDRPlay all other devices are incorrect.

scancapecod, 2025-03-22:
That's what I thought, and it seems to be the general consensus/gripe from all other third party software writers that are supporting it. It is definitely something that can be overcome, but anyone using the SDRPlay with this, or any third party software, needs to be aware of it. 

No need to keep this one open for my sake. Thanks.

paulh002, 2025-03-22:
What I can do is make a configuration item in sdrberry_settings to invert this. There is a discussion in the SoapySDRPlay repo in this but it is from 2021 so it looks like they don't want to change it.

scancapecod, 2025-03-22:
Happy to test for you when you have the time. Otherwise no issue, I just slide 'em around until I get optimum reception, and go from there! :-)