paulh002/sdrberry #46: Problem with entering frequency

MDHebert, 2025-03-25:
Great implementation Paul,
I have a pi4 on the older Raspberry pi 7" screen, everything is perfect except, I can enter in a frequency, but hitting the return button does not transfer the typed number to a vfo, so I'm unable to change the freq. I can switch bands, everything else is fine, tried with mouse as well   Thanks

paulh002, 2025-03-25:
Are you using onscreen keyboard or real keyboard. On screen type for
example 14.2 MHz then return
Real keyboard 14.2 m

MDHebert, 2025-03-25:
wow, such a quick response. Didn't think of trying keyboard, but same result, even using mouse. Return button just clears out the dialog entry. Its an SDRplay as well. Thanks

paulh002, 2025-03-25:
The frequency should be valid for ham band. But I will check it later, and
let you know

MDHebert, 2025-03-25:
I think I''ve found the issue. I'm not able to enter freq. outside of ham bands with the limit vfo to ham bands un checked or not. I usually check operation with the local AM station that I can see the tower of. Thanks again for the prompt response. I'm trying to set up a simple system for my makerspace to explain some theory to.

paulh002, 2025-03-25:
You can add the am to the cfg file as a band. But I will update it that the
check will be skipped when the checkbox is off

closed 2025-03-27 by MDHebert