paulh002/sdrberry #52: Can't tune to ...anything

phobosd, 2025-03-30:
Hello! I just finished installing SDRBerry on a fresh pi4 64bit OS - compiled it for RTLSDR and the DSI screen. It starts up but I'm unable to tune to anything. In the photo I'm simply trying to tune to a local FM station. The waterfall is also going crazy...

Here are my rtlsdr settings in sdrberry_settings.cfg:

[rtlsdr]
audiobuffersize = "4096"
correction = "2"
decimate = "0"
gain = "40"
if-gain = 41
resamplerate = "0.023531"
rf-gain = 10
samplerate = "2048"
span = 2048

Perhaps something in there is incorrect? I followed a Tech Minds video to get this all set up. 

Also can you recommend a good SDR Discord I could join to perhaps help?

Appreciate your work on this project! It's very cool.

[image]

paulh002, 2025-03-30:
You have to reduce the gain until the slider is not orange any more.
Also select bFM as mode, because FM band is broadband mode.

phobosd, 2025-03-30:
Sound is also very scratchy - might be beacuse of my tuning problem, but the console is scrolling this:

phobosd, 2025-03-30:
Sound is also off...the log is scrolling a lot of this:

Input device name: "vc4-hdmi-0"
Input device name: "vc4-hdmi-0 HDMI Jack"
Input device name: "vc4-hdmi-1"
Input device name: "vc4-hdmi-1 HDMI Jack"
Input device name: "10-0038 generic ft5x06 (00)"
Input device name: "vc4-hdmi-0"
Input device name: "vc4-hdmi-0 HDMI Jack"

I'm using this display: https://www.amazon.com/dp/B0D3QB7X4Z?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1

Also i've tried all the modes, I think something else is off...

phobosd, 2025-03-30:
[image]

Bandwith set to 0 Khz seems not ideal?

paulh002, 2025-03-30:
> image.png (view on web)
>
> Bandwith set to 0 Khz seems not ideal?

Rtlsdr does not have bandwidth filter if I remember correct

paulh002, 2025-03-30:
Please check first the wiki for tips

n1ai, 2025-04-02:
_Perhaps something in there is incorrect? I followed a Tech Minds video to get this all set up._

Did you try using a USB audio device just like Tech Minds did?  I have asked the author and he says it is required, see Discussions tab here on github for more info.

phobosd, 2025-04-02:
yup i sure did. i think my problem is moreso how the software is working with the RTL SDR more than anything. even with the USB device i'm still getting all static.

paulh002, 2025-04-03:
You can try the latest build version. 
And set the config for rtlsdr like this in sdrberry_settings.cfg
```
[rtlsdr]
audiobuffersize = "8192"
decimate = "0"
gain = "10"
if-gain = 10
samplerate = "2048"
span = 2048
```

phobosd, 2025-04-03:
Is there a more recent release than July?

I used this script on March 31st to install:
https://raw.githubusercontent.com/paulh002/sdrberry/master/install/install.sh

phobosd, 2025-04-03:
I've attached logs of a session. This is me starting sdrberry, turning to a local FM station, and then setting the mode to bFM.

Hopefully it helps, or shows where I might be configuring the interface incorrectly to simply tune in an FM station?

Option name: bass value: 4
Option name: fast value:
Option name: medium value:
Option name: mode value: 0
Option name: ratio value: 12
Option name: slow value:
Option name: threshold value: 0
Option name: treble value: 0
Screen resolution 800 x 480 screenRotate 0 touch_swap setting 0
Input device name: "C-Media Electronics Inc. USB Audio Device"
Device is not a keyboard
Input device name: "vc4-hdmi-0"
Device is not a keyboard
Input device name: "vc4-hdmi-0 HDMI Jack"
Device is not a keyboard
Input device name: "vc4-hdmi-1"
Device is not a keyboard
Input device name: "vc4-hdmi-1 HDMI Jack"
Device is not a keyboard
Input device name: "10-0038 generic ft5x06 (79)"
Device is not a keyboard
end keyboard
Input device name: "C-Media Electronics Inc. USB Audio Device"
Input device name: "10-0038 generic ft5x06 (79)"
Input device name: "vc4-hdmi-1 HDMI Jack"
Input device name: "vc4-hdmi-1"
Input device name: "vc4-hdmi-0 HDMI Jack"
Input device name: "vc4-hdmi-0"

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi0,0), Unknown error 524.

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi1,0), Unknown error 524.

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi0,0), Unknown error 524.

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi1,0), Unknown error 524.

129 device: Default ALSA Device input 32 output 32
130 device: PulseAudio Sound Server input 32 output 32
131 device: bcm2835 Headphones (bcm2835 Headphones) input 0 output 8
132 device: USB Audio Device (USB Audio) input 1 output 2
audio input device = 132 USB Audio Device (USB Audio) samplerate 48000 channels 1

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi0,0), Unknown error 524.

RtApiAlsa::probeDeviceInfo: snd_pcm_open (playback) error for device (hw:vc4hdmi1,0), Unknown error 524.

device name = Default ALSA Device
: maximum output channels = 32
device name = PulseAudio Sound Server
: maximum output channels = 32
device name = bcm2835 Headphones (bcm2835 Headphones)
: maximum output channels = 8
device name = USB Audio Device (USB Audio)
: maximum output channels = 2
audio device = 132 USB Audio Device (USB Audio) samplerate 48000 channels 2
Cannot connect to 56
Cannot connect to 57
Cannot connect to 32
Cannot connect to 33
Cannot connect to 63
The framebuffer device was opened successfully.
800x480, 16bpp
The framebuffer device was mapped to memory successfully.
Input device name: "C-Media Electronics Inc. USB Audio Device"
Input device name: "vc4-hdmi-0"
[... 156 lines cut ...]
[DEBUG] Setting center freq: 98900000
[R82XX] PLL not locked!
3 m i2c  I2C time 2
select_mode_rx start rx threads
iir filter [sos]:
iir filter | sos:
  b :   0.06705825,  0.13411650,  0.06705825
  a :   1.00000000, -1.05075324,  0.28519174
iir filter | sos:
  b :   0.06705825,  0.13411650,  0.06705825
  a :   1.00000000, -1.16203701,  0.42130458
iir filter | sos:
  b :   0.06705825,  0.13411650,  0.06705825
  a :   1.00000000, -1.42308676,  0.74059838
Buffer queue 0 Radio samples 65536 Audio Samples 3072 Passes 0 Queued Audio Samples 2048 droppedframes 0 underrun 3
rms 0.000038 db -88.500534 envelope 0.010617
Buffer queue 1 Radio samples 65536 Audio Samples 3072 Passes 0 Queued Audio Samples 2048 droppedframes 0 underrun 3
rms 0.000038 db -88.447847 envelope 0.010639
Buffer queue 1 Radio samples 65536 Audio Samples 3072 Passes 0 Queued Audio Samples 2048 droppedframes 0 underrun 0
rms 0.000038 db -88.453088 envelope 0.010664
Buffer queue 0 Radio samples 65536 Audio Samples 3072 Passes 0 Queued Audio Samples 2048 droppedframes 0 underrun 0
rms 0.000038 db -88.456518 envelope 0.010632

Caught Ctrl+C (SIGINT), exiting gracefully...
terminate called without an active exception

paulh002, 2025-04-05:
I cannot judge the log, on first look it looks ok. Unfortunately I don't own a V4 RTLSDR only a v3 and that one works good with me. I just released a new version maybe you can try that.
One observation is the message that the PLL is not locked. That looks a bit strange.

n1ai, 2025-04-05:
> One observation is the message that the PLL is not locked. That looks a bit strange. I'd guess you specified a frequency that's outside of the range of the RTLSDR. That would cause the PLL to not lock. Whether this is user error or a driver problem, I couldn't say.

It is strange, but also very common.  I've seen it for at least five years now.  A bit of googling says it happens because the software starts accessing the driver before the PLL is locked, but it does lock pretty quickly thereafter so things typically work as expected.  

>  Unfortunately I don't own a V4 RTLSDR only a v3 and that one works good with me. 

My experience using rtl-sdr v4 on debian 12 (bookworm) was that the files you need from the repo i.e. librtlsdr-dev, work for the v3 but didn't have the v4 code.   In theory the v4 should just act like a v3 but you won't get the benefit of the main feature, the fact it has a built-in up-conveter to cover the HF bans.  

Also in theory you should be able to follow the steps in https://www.rtl-sdr.com/v4/ to get a fully functioning rtl-sdr v4 on debian.    I don't know if this approach works correctly with sdrberry though.

Typically I build most of the Linux SDR software stack I want to use ( gnuradio, soapy, gqrx, etc ) myself from source using a script I wrote.  This is because while Debian has gotten better at keeping the SDR bits at a decently up-to-date level, sooner or later I find something I care about is out of date and then I find I have to build it all from scratch to get all these things to work together correctly.  Unfortunately,  I haven't integrated sdrberry into this scripting yet.    I am just using Paul's script and a fresh OS install for sdrberry.

> I used this script on March 31st to install

The date of the script isn't important, it just builds the current main branch.

The commit history ( https://github.com/paulh002/sdrberry/commits/master/ ) shows there have been some changes since March 31st on the main branch, and some are related to rtlsdr.

phobosd, 2025-04-13:
i watched your video that you recently put out, so i decided to try this again. with a fresh install, starting from step 1 like you in the video. this time i'm using an RSPDx R2. i noticed that you don't have this output when building SDRberry. i'm wondering if i'm missing a step or a dependency?

This scrolls quite a bit during the ./install.sh SDB DSI

/home/pi/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx: In function ‘PyObject* _wrap_SoapySDRDoubleList_swap(PyOb                                                                                              ject*, PyObject*)’:
/home/pi/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx:33241:62: warning: unused parameter ‘self’ [-Wunused-param                                                                                              eter]
33241 | SWIGINTERN PyObject *_wrap_SoapySDRDoubleList_swap(PyObject *self, PyObject *args) {
      |                                                    ~~~~~~~~~~^~~~
/home/pi/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx: In function ‘PyObject* _wrap_SoapySDRDoubleList_begin(PyO                                                                                              bject*, PyObject*)’:
/home/pi/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx:33305:63: warning: unused parameter ‘self’ [-Wunused-param                                                                                              eter]
33305 | SWIGINTERN PyObject *_wrap_SoapySDRDoubleList_begin(PyObject *self, PyObject *args) {
      |                                                     ~~~~~~~~~~^~~~
/home/pi/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx: In function ‘PyObject* _wrap_SoapySDRDoubleList_end(PyObj                                                                                              ect*, PyObject*)’:
/home/pi/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx:33361:61: warning: unused parameter ‘self’ [-Wunused-param                                                                                              eter]
33361 | SWIGINTERN PyObject *_wrap_SoapySDRDoubleList_end(PyObject *self, PyObject *args) {
      |                                                   ~~~~~~~~~~^~~~
/home/pi/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx: In function ‘PyObject* _wrap_SoapySDRDoubleList_rbegin(Py                                                                                              Object*, PyObject*)’:
/home/pi/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx:33417:64: warning: unused parameter ‘self’ [-Wunused-param                                                                                              eter]
33417 | SWIGINTERN PyObject *_wrap_SoapySDRDoubleList_rbegin(PyObject *self, PyObject *args) {
      |                                                      ~~~~~~~~~~^~~~
/home/pi/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx: In function ‘PyObject* _wrap_SoapySDRDoubleList_rend(PyOb                                                                                              ject*, PyObject*)’:
/home/pi/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx:33473:62: warning: unused parameter ‘self’ [-Wunused-param                                                                                              eter]
33473 | SWIGINTERN PyObject *_wrap_SoapySDRDoubleList_rend(PyObject *self, PyObject *args) {

phobosd, 2025-04-13:
well i guess it doesn't matter because i got it working with the RSPDX R2! just can't change frequency by touch/dragging. not sure if that's a feature.

paulh002, 2025-04-13:
This scrolls quite a bit during the ./install.sh SDB DSI

should be ./install SDP DSI

paulh002, 2025-04-13:
Yes, touch dragging does not work. Maybe i add it later, but it is not so
accurate.

paulh002, 2025-04-14:
If you take the latest version you can also change frequency by dragging
using the touch screen.

73,
Paul

phobosd, 2025-04-14:
how would i go about "updating" it? using ./install.sh SDP DSI again?

and great! i'll try it out :)

paulh002, 2025-04-14:
cd sdrberry
git pull
cd build
cmake ..
make -j4
sudo make install

phobosd, 2025-04-14:
thank you! is it possible to build sdrberry for both sdrplay and RTLSDR? i'd like to interchange them every now and then.

do i just modify the sdrberry_settings.cfg file before starting?

paulh002, 2025-04-14:
Yes, that is easy. Only the install script can only do one at the time. But
you can add any sdr manually. If you check the install script you can just
execute the part of the sdr you want to add. If you dont have the knowledge
i will add some scripts for adding a sdr

paulh002, 2025-04-14:
You can switch direct in sdrberry if the SoapySDR drivers are available