paulh002/sdrberry #57: Add bands?

radiogareth, 2025-04-08:
Not an issue as such, more a wish-list.....can we have bands up to 6Ghz, or even user-defined bands. I have odd IFs on some of my transverters which would be useful. Pluto and HackRF user mainly, but also the cheaper "Software defined radio SDR AD9363 AD9361 ZYNQ7020 ADI Pluto communication experiment platform" as found on AliExpress for around £125ish.
Thanks
Gareth

paulh002, 2025-04-08:
Hi Gareth, yes you can easily add bands in sdrberry_settings.cfg file. The
band's show depends on the SoapySDR driver response. But if necessary it
can be overwritten. I will make a wiki page to explain how to do it. 73 Paul

paulh002, 2025-04-08:
You can check this wiki page https://github.com/paulh002/sdrberry/wiki/band-configuration

closed 2025-04-25 by paulh002