paulh002/sdrberry #58: Physical PTT button for tx

Mavik7zip, 2025-04-08:
is there a way to use a physical ptt button to go into transmission with raspberry pi 4 and hackrf? I saw that on paulh002's github there is the ESP32Controler but even there I didn't see any provision for a physical ptt.
also I wanted to know if there is any provision for rf switches to be able to use an external power amplifier.
thanks in advance, 73

paulh002, 2025-04-08:
In the ESP32 the IO marked with rxtx is the hardware port for rx/tx control. You can also connect a optical rotary controller to the ESP32. The ESP32 will send a CAT command to switch between rx and tx or to set the frequency.
In sdrberry you can setup I2C interface to switch a bandpass filter and / or amplifier. 
At the moment I don't have a direct input on the pi, but that should not be such a difficult task to add. An option could also be to use I2C to readout a PCF8574(A).
In principle you can use any arduino with a usb connection to send CAT commands. I use the FT891 cat commands. Although I added several commands to have more functionality.

radiogareth, 2025-04-09:
Hi Paul, It may be worth having a look at both the BATC Portsdown 4 project and the Langstone by Colin G4EML.
https://wiki.batc.org.uk/Portsdown_4 and https://github.com/g4eml/Langstone 
They both have PTT direct to the RPI and the Langstone also has a Band expansion capability which may be of use as your project expands. I use both with a Pluto but other than burning SD cards and a few alterations to config I'm not competent at anything Linux.
Gareth

paulh002, 2025-04-09:
I prefer to use I2C connectivity it is relative simple and small IO boards are available on ebay or aliexpress.  Also the radioberry SDR uses most GPIO ports and I don't want to have conflicts with this sdr board (it is my main sdr for hf). 
A nice option is the ABE_ExpanderPi. It has multiple IO's and also analog IO's. I use it for my next project a pi ldmos controller.

[image]

Mavik7zip, 2025-04-10:
> In the ESP32 the IO marked with rxtx is the hardware port for rx/tx control. You can also connect a optical rotary controller to the ESP32. The ESP32 will send a CAT command to switch between rx and [...]

HI in these 2 days I tried to write the code of the esp32controller on my esp32s3 without success I tried with arduino ide platformio and visual studio 2022 and in all I encountered problems that I could not solve. what is the correct way to write the code on the microcontroller? is there a guide to follow?

paulh002, 2025-04-10:
I don't think it will work on EPS32S3, the repo is for ESP32. It is normal arduino code. But it will need a number of libraries. Because the repo is older, you need to correct versions of lvgl installed (v8.3.3).  Also there is a file missing, I added the file.
You can use the ESP32S3 version, I just made that repo public you can try. Use the Arduino IDE, for visual studio you need a plugin "visual micro" which is a payed tool. Note for the ESP32 S3 the IO ports config is different.

paulh002, 2025-04-10:
For your information I am adding a I2c input interface for RX/TX switching and for some other buttons. Will take a few days to finish. Maybe it is more easy. You can use a cheap PFC8574 interface module and connect it to the I2c pins of the pi.

Mavik7zip, 2025-04-11:
I'm having trouble installing all the libraries on the Arduino IDE, in particular I can't find the "network.h" library. It would be a good thing if you could make a single package with the libraries too

> I don't think it will work on EPS32S3, the repo is for ESP32. It is normal arduino code. But it will need a number of libraries. Because the repo is older, you need to correct versions of lvgl [...]

paulh002, 2025-04-11:
Ok, will update later tonight

Mavik7zip, 2025-04-11:
ok thanks, anyway I decided to wait for the version with PFC8574 I don't want to give you more "useless" work if you are already working on a better version.
thanks for everything I'm waiting for the new version

> Ok, will update later tonight
> 
> Op vr 11 apr. 2025 20:26 schreef Gabriele Vallosio ***@***.***
> […](#)

paulh002, 2025-04-11:
I made ESP32S3Controler private again, it needs some rework because multiple arduino libraries are changed. This is something I will do later. Now I want to finish the I2c option. I will update you when you can try it.

paulh002, 2025-04-12:
If you like you can try the build version, I added pcf8574 support. (Other devices can be selected but not yet tested). GPIO 0 is the rx/tx switch. If you ground it TX is active. I only implemented rx/tx switch now. The other ports still have to do something with it. Or push buttons or rotary controler for volume and gain or so. Rotary for tuning is maybe not fast enough but have to check.
I use a cheap board from amazon to test it. And it looks ok.

```
cd sdrberry
git pull
git checkout build
cd build
cmake ..
make -j4
sudo make install
```

[image]

Mavik7zip, 2025-04-23:
hello this evening I finally managed to put all the pieces together, and I can happily say that everything works. my current setup is with the pcf8574 via i2c and it works very well I only had to enable the i2c via raspi-config. paulh002 I want to compliment you on the fantastic project and the speed with which you fixed this problem of mine.
in addition I wanted to ask you how you see the software support for using an sdr in reception and another sdr for transmission, for example SDRPlay RSP1A (14bit adc) in reception and Hackrf (10bit tx) one in transmission. congratulations again and good continuation.
73

paulh002, 2025-04-25:
Thanks, good it works now. Switching SDR between rx and tx is a good idea, but requires some work. Maybe later.

closed 2025-04-25 by Mavik7zip