paulh002/sdrberry #59: Had SDRBerry running perfectly, but then I modified the bands.

Bob-C-UK, 2025-04-16:
First of all, thank you for you excellent work.

I had SDRBerry 100% working  - when I eventually had an eureka moment and worked out to turn the screen up the other way as I could not find a software method the reverse the mouse movement which was move mouse right, pointer moved left,  mouse move left, pointer moved right and Y axis also backwards.
This was all working with an Adafruit Speaker Bonnet featuring a max98357a chipset.
Then I made some modification to the sdrberry_settings.cfg for bands, having previously been successful in adding the civil air band frequencies. I do not know what happened but the sdrberry screen would freeze on selecting a band. I could not see an error on inspection of the file so I took what I thought was the easy way out and started over with a fresh OS install on a new 64GB Samsung Fit Plus drive.
Since then nothing has worked and on start up I am left with Segmentation Fault that seems to have something to do with Alsa and pulse_connect audio.

My information as follows:

Raspberry Pi 5 with official 7 inch Touch Screen 2.
Raspberry Pi OS 64 bit
Installation of OS using official Raspberry Pi Imager
Modification of raspi-config to give autologon to console.
Installation of SDRPlay API
SDRBerry install using  ./install.sh SDP T2
Please see attached files for modifications of cmdlines.txt, config.txt, sdrberry_settings.cfg

[cmdline.txt](https://github.com/user-attachments/files/19767285/cmdline.txt)

[config.txt](https://github.com/user-attachments/files/19767288/config.txt)

[sdrberry_settings.cfg.txt](https://github.com/user-attachments/files/19767291/sdrberry_settings.cfg.txt)

[log.txt](https://github.com/user-attachments/files/19767381/log.txt)

[sdrberry.log](https://github.com/user-attachments/files/19767383/sdrberry.log)

Would you have any ideas as to what the segmentation fault might be ? I have tried a second new install on the new drive and same results. Going round in circles it seems.

Still a novice in Linux

Many thanks in advance.

Regards

Bob.

paulh002, 2025-04-16:
Hi Bob, Please when installing do not update cmdline.txt for rotating the screen. That is done by sdrberry self. During installation of Touch 2 display the screen and mouse is already rotated. So before installation only update the pi to the latest version (sudo apt update and sudo apt upgrade).
At the moment sdrberry does not support audio output only devices. So the max98357a will not be recognised because it does not have audio input. Please use (for now) an USB audio adapter. Alsa default and pulse audio is also not supported. In the latest version it is not in the list any more.

paulh002, 2025-04-16:
Hi Bob,
I updated sdrberry to support audio output only devices. If you install the build version you can try it.
Install like this:
```
./install.sh SDP T2 build
```

do1tl, 2025-04-16:
Hi Paul

Does ./install.sh SDP DSI build also work?

Sorry for my poor English.

I've been trying to build an SDRBerry for days.

And I'd like to use it as a pan adapter for my Kenwood TS-480.

Thank you for this project.

paulh002, 2025-04-16:
Yes, that also works

Bob-C-UK, 2025-04-16:
Paul,
Many thanks for your immediate response this morning and also for updating SDRBerry to allow for audio output only devices.
The build install has worked with the Adafruit Bonnet board. It is great to see it all working again.
This afternoon I ordered a Waveshare USB sound dongle. Have to wait until tomorrow for it to arrive. However I am happy with the working results using the max98357a Adafruit device.
Thanks once again for your excellent work and attention.
Here is a picture screen build.

[image]

Regards / 73

Bob.

paulh002, 2025-04-16:
Looks good. Happy that it works for you. 

73,
Paul

closed 2025-04-25 by paulh002