paulh002/sdrberry #60: Raspberry pi 3b+

do1tl, 2025-04-17:
Hi Paul

Does this only work on RPi 4 and 5?

I have an RPi 3b+ and I can't even get the program to start.

I type sudo sdrberry > sdrberry.log 2>&1 but nothing happens, not even an error message.

I use Bullseye.

Greetings

paulh002, 2025-04-17:
Pi3 is much to slow. I never tested on pi3

do1tl, 2025-04-17:
Then I'll get an RPI 5.

I love this project.

I thought I'd give it a try, but unfortunately the program won't start.

do1tl, 2025-04-17:
Okay, thank you
for the help.

There's a video on YouTube with an Rpi400 :-)

I'm getting an Rpi 5 4gb

I'll post pictures as soon as I'm done.

do1tl, 2025-04-17:
Hi Paul

I finally got it working on the Raspberry Pi 3b+, and it's not bad at all. I'll continue this afternoon and post pictures and videos then.

Hardware:
Raspberry Pi 3B+
SDRPlay RSP1 (MSI001)
Nooelec Nesdr Smart (RTLSDR)

Regards, DO1TL

paulh002, 2025-04-18:
Unbelievable :)

do1tl, 2025-04-18:
Okay, I'm waiting for my Rpi 5.

The system is a bit weak for a pan adapter.

Greetings

closed 2025-04-25 by paulh002