paulh002/sdrberry #61: Display doubled and jumbled when operating ui

djongers, 2025-04-21:
[image]
[image]

display seems to be doing odd things

Original rpi 7inch display on raspberry pi 5

pi@sdrberry:~ $ fbset

mode "800x480"
    geometry 800 480 800 480 32
    timings 0 0 0 0 0 0 0
    rgba 8/16,8/8,8/0,0/0
endmode

Any ideas?

paulh002, 2025-04-21:
For which display did you install?
Is this the old 7 inch pi display, or the new Touch2 display?

djongers, 2025-04-21:
the old 7inch display

paulh002, 2025-04-21:
How did you install? with DSI or T2?

djongers, 2025-04-21:
DSI and RTL

paulh002, 2025-04-21:
I don't have an install option for the T1 display at the moment. I will add that later. But use the T2 option instead of DSI and manually set the resolution to 800x480 in sdrberry_settings.cfg.
```
[screen]
resolution = "0"
rotation = "0"
```

djongers, 2025-04-22:
appears to have the same result once the settings are changed.

paulh002, 2025-04-22:
Hi, I made some updates to instllation script and the software. Please install using T1 display option. SO if you are using SDRPlay it should look like this:
```
./install.sh SDP T1
```

If it does not work, please let me know what operating system did you use? Did you update the operating system and share the config.txt and cmdline.txt

djongers, 2025-04-23:
downloaed updated script excuted with RTL & T1 see result below seems to be doubled in 2/3 of the screen then 1/3 blank

[image]

used lite raspbian

pi@sdrberry:~ $ uname -a
Linux sdrberry 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux

 config.txt 

 For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtparam=nvme
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
#dtoverlay=vc4-kms-v3d
max_framebuffers=2
framebuffer_depth=16
# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan

disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]

cmdline.txt

console=serial0,115200 console=tty1 root=PARTUUID=46d705ab-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=NZ vt.global_cursor_default=0 usbhid.mousepoll=2 vt.global_cursor_default=0 usbhid.mousepoll=2 vt.global_cubal_cursor_default=0 usbhid.mousepoll=2 vt.global_cursor_default=0 usbhid.mousepoll=2 vt.global_cursor_default=0 usbhid.mousepoll=2 vt.global_cursor_default=0 usbhid.mousepoll=2 vt.global_cursor_default=0 usbhid.mousepoll=2

djongers, 2025-04-24:
swapped to a raspberry pi 4 and have got it going

closed 2025-04-24 by djongers