paulh002/sdrberry #62: waveshare 7 inch touch display

Mavik7zip, 2025-05-05:
Hi Paulh.I'm having trouble getting this 7 inch waveshare touch display to work. The main problem I think is that it's not recognized as a keyboard. The name it's recognized by is ```
Input device name: "wch.cn USB2IIC_CTP_CONTROL"
Device is not a keyboard
```

as always thanks in advance

paulh002, 2025-05-06:
You can try to edit sdrberry_settings.cfg

[input]
mousexx = "Mouse"
touchscreen = "wch.cn USB2IIC_CTP_CONTROL"
touch_swap_xy = 0
touch_debug = 0

Mavik7zip, 2025-05-06:
hi, I inserted what you wrote to me in the configuration file and now the touch is read but now I encounter a problem with the touch mapping. I saw this display with the touch debug that returns data from 0 to 2034 on the x and 0 to 2024 on the y and the software interprets the data as if the touch coordinates were 1 to 1 with the resolution (1024x600)

this is a tap on the bottom right of the display
`touch x: 2034 y 2024 hor_res 1024 ver_res 600
touch x: 1023 y 599 hor_res 1024 ver_res 600`

paulh002, 2025-05-06:
you can try to enable EVDEV_CALIBRATE in the lv_drv_conf.h file, here a mapping can be enabled and set.

[image]

set the EVDEV_CALIBRATE to 1 and define the mapping values.

Mavik7zip, 2025-05-06:
I did the manual mapping I recompiled and now it started working very well thank you for this. I wanted to ask you if you have updates for the physical vfo with the pcf8574.
thanks for everything have a good continuation.
73

closed 2025-05-11 by Mavik7zip