paulh002/sdrberry #63: Hackrf Tx power

Mavik7zip, 2025-05-09:
hello in these days I managed to refine my setup with hackrf and amplifier able to offer with an input of 15dbm 25w but after a few tests I realized that there was something wrong with the output power of the amplifier, then I started to investigate, since I do not have great equipment to understand in broad terms the power with which I was coming out I put an ammeter to see the absorption of the amplifier knowing that it should absorb a maximum of 1.2A to give an output of 25w. setting the drive to 61 from the TX menu the absorption of the amplifier is 450ma therefore an output of approximately 4.8w doing numerous tests I realized that if during the transmission I changed the drive the absorption rose to 1.4A (29.9W) but once the ptt was released and pressed again the absorption returned to 450ma.
kind regards

https://github.com/user-attachments/assets/5ab78d0c-f7d5-4c58-aaa8-431d9b48f924

paulh002, 2025-05-10:
Ok, will check it. Maybe the hackrf need drive setup for every transmit.
This depends a bit per SDR.

73,
Paul

paulh002, 2025-05-10:
You can update sdrberry_settings.cfg
set restart_tx = "1"

I also made some updates to sdrberry, you can also update to the latest version. But the settings change should already fix the issue.

[hackrf]
audiobuffersize = "4096"
bandwidth = "0"
dc = "0"
decimate = "4"
drive = "61"
halfduplex = "1"
if-gain = 21
resample_rate = {"0.192672", "4000", "4"}
restart_tx = "1"
rf-gain = 10
samplerate = "4000"
samplerate_tx = "4000"
span = 250

Mavik7zip, 2025-05-11:
ok thanks paulh I updated the config without updating all the software but it already works perfectly, thanks as always, always very fast in responding.
73

closed 2025-05-11 by Mavik7zip

paulh002, 2025-05-11:
Thanks, I just fix one other issue, related to TX buttons not available, this was caused by an issue in the selection of USB audio devices. If you experience this you can update to the latest version.