paulh002/sdrberry #67: choppy tx audio after a few reboots

Mavik7zip, 2025-05-24:
hi paulh, after using the software for a while I noticed that periodically I have a problem with the transmission I don't think it's an audio problem since the receiving radio that I use for these tests detects a very inconsistent signal from s4 to s9+40 and consequently a crackling audio. I'm using a setup with an amplifier switch rf etc that works perfectly in the first starts of the software. I also noticed that this problem is triggered in the abrupt closing of the software for example when you terminate the software by pressing ctrl+c in the terminal 2 times by mistake, I also wanted your confirmation if this is the correct way to terminate the software. now I'll do a clean installation of the software with the latest version and check if over time this problem comes out again. bye

https://github.com/user-attachments/assets/eb28a6f1-104c-4fb3-ad99-e81f1395f7d8

paulh002, 2025-05-24:
Can you send the log file of the startup when this happens, I did several changes to the audio usb driver selection. There we some issues. Try it with the latest version. 
But please send the log of the faulty session, may it shows something during startup. Anyway I will also test it.

paulh002, 2025-05-24:
And yes, there is no official shutdown button. But ctrl C is ok. Should not give an issue. I use the powerbutton on my tranceiver.

Mavik7zip, 2025-05-24:
ok i have some RtApiAlsa error, now I proceed by installing the latest version..

[log.txt](https://github.com/user-attachments/files/20426544/log.txt)

closed 2025-05-25 by paulh002

Mavik7zip, 2025-05-27:
hi paulh unfortunately the problem has reappeared in the same form plus in this new version I noticed that if I set the drive power less than or equal to 20 it seems that the lna is disabled

https://github.com/user-attachments/assets/85e59fb2-1305-4b0a-8f8e-1c620b5cfc70

paulh002, 2025-05-27:
On the lna behaviour, the hackrf is a simplex device. So it cannot transmit and receive at the same time. I think this is behaviour of the hackrf it self. I just send gain command to the SoapyHackRF driver. 
If you check the block diagram you can see there is a shared path for rx and tx where a 2 amplifiers are situated. Maybe this is the reason why it behaves like that.

[image]

paulh002, 2025-05-28:
I did some more testing on hackrf, I also see that the transmit power is changing when changing the receive gain. Do you also have this behaviour? What version of firmware are you using in the hackrf?

Mavik7zip, 2025-05-28:
hi, i have not noticed the variation of the tx power when the gain varies, every now and then i have noticed variations in the amplifier absorption but i have always blamed my antenna setup not well calibrated and my trx switch not optimal, this evening check this too. tonight i will try if on the old version the gain varies if i lower the tx power below 20. then i will check if on the new version if i select the usb audio card again the transmission is fixed. as for the hackrf version it is a clifford R9 i do not know the firmware version and i have never worried about it, if you advise me to update it i will update it.

paulh002, 2025-05-28:
> :
>
> I found in a github issue what the 116db gain in reception and the 61 in
> transmission correspond to and I noticed that there is AMP Gain of 14db in
> common and it has a step of 14 so on and off a while ago I did a bit of
> research to understand why on soapysdr the gain is 116db and on sdrpp I had
> distinctly the LNA 0-40 step of 8 the VGA 0-62 step 2
> and an amp enabled on/off voice and on soapysdr they are compiled to have
> 0-116 in steps of 1
> ------------------------------
> -- RX Channel 0
>
> Full-duplex: NO
> Supports AGC: NO
> Stream formats: CS8, CS16, CF32, CF64
> Native format: CS8 [full-scale=128]
> Stream args:
> * Buffer Count - Number of buffers per read.
> [key=buffers, units=buffers, default=15, type=int]
> Antennas: TX/RX
> Full gain range: [0, 116] dB
> LNA gain range: [0, 40, 8] dB
> AMP gain range: [0, 14, 14] dB
> VGA gain range: [0, 62, 2] dB
> Full freq range: [0, 7250] MHz
> RF freq range: [0, 7250] MHz
> Sample rates: 1, 2, 3, 4, 5, ..., 16, 17, 18, 19, 20 MSps
> Filter bandwidths: 1.75, 2.5, 3.5, 5, 5.5, ..., 14, 15, 20, 24, 28 MHz
> ------------------------------
> -- TX Channel 0
>
> Full-duplex: NO
> Supports AGC: NO
> Stream formats: CS8, CS16, CF32, CF64
> Native format: CS8 [full-scale=128]
> Stream args:
> * Buffer Count - Number of buffers per read.
> [key=buffers, units=buffers, default=15, type=int]
> Antennas: TX/RX
> Full gain range: [0, 61] dB
> VGA gain range: [0, 47, 1] dB
> AMP gain range: [0, 14, 14] dB
> Full freq range: [0, 7250] MHz
> RF freq range: [0, 7250] MHz
> Sample rates: 1, 2, 3, 4, 5, ..., 16, 17, 18, 19, 20 MSps
> Filter bandwidths: 1.75, 2.5, 3.5, 5, 5.5, ..., 14, 15, 20, 24, 28 MHz
>
> image.png (view on web)

I have checked the source code for soapyhackrf and there are some rules for
the gain if you use a linear gain like I do. Half way they switch on the
amp and deduct 14db from the VGA gain. The same for tx. The amp on/off
switch is shared by Rx and tx, that explains the RX gain influence. In
fortunate it looks like my hackrf amp is broken, that's why I have the
strange effect. In general it would be better to operate the amp
separately. But you don't want specific code for each sdr. However you can
access the amp separately through SoapySDR. But i need some methods to
detect this. I will try something later on.

Mavik7zip, 2025-05-28:
I did the tests and I can confirm that the reception gain affects the transmission, the transmission problem is not solved by selecting the USB card now I'm having strange problems with the transmission but I think it's a problem with my setup in a few words I select the 2m band I press tx and my raspberry turns off. in the old one (I kept the sd with the installation that had a problem at the beginning) the reception gain does not change based on the tx power and not even the tx power based on the reception gain. in the following days I will redo the installation with the new version to make sure that the problems I'm having are not inherent to the setup (if you tell me that you will release a new version soon I'll wait) 
I attach the boot log of the new version

[bootlog.txt](https://github.com/user-attachments/files/20494285/bootlog.txt)

Mavik7zip, 2025-05-29:
is this the error related to pulseaudio?
`ALSA lib control.c:1528:(snd_ctl_open_noupdate) Invalid CTL pulse`

paulh002, 2025-05-29:
> :
>
> This evening I turned on the raspberry and after 5 minutes that it was on
> it became a smoking machine so for a while I won't be able to do the tests
>
> 2.jpeg (view on web)
> 1.jpeg (view on web)

That's bad luck, what happend? Power adapter failure? I am adding some
stuff for managing the gain more individual. That will take some time.
Hopefully you will be able to replace it later on.

paulh002, 2025-05-29:
> :
>
> I think it broke because the component gave up I power it with a good
> stepdown given for 400w so I see it difficult to understand if it is for
> ripple and the voltage is set to 5.08v the range is 4.9-5.25v. What do you
> recommend I take a 4gb pi4 or can the 2gb be enough?

I use both 2gb and 4gb if you only run sdrberry 2gb is ok.

Mavik7zip, 2025-06-28:
Hi paulh im back with new hardware have you made any progress with the developments is there anything to test regarding the audio problems i was having?

paulh002, 2025-06-28:
You installed already the latest version? I fixed some hackrf issues and other bugs, but Not specifically for audio. Best just to do a general check and start from there. I was busy with some other stuff, and general issues.

Mavik7zip, 2025-06-29:
hi paulh today i did some radio listening but i had to stop after a while because after a few tens of minutes of use the waterfall has a delay of several tens of seconds so adjusting the gain and changing frequency becomes very slow and unusable, however i noticed some differences with the latest firmware in practice when i change frequency the gain goes to 0 and sometimes it returns to the preset value other times it stays at 0. i also recovered a small endfed in this period so in these days i will update you on the transmission and transmitted audio.
bye

paulh002, 2025-06-29:
This is with hackrf? What sample rate did you use? You can config sample rate correction, but share a log. Then i can test also

paulh002, 2025-06-29:
> rx 4000Khz 8 decimate 1750khz bandwidth 192khz span
> tx 4000khz
>
> Option name: bass value: 4
> Option name: fast value:
> Option name: medium value:
> Option name: mode value: 0
> Option name: ratio value: 12
> Option name: slow value:
> Option name: threshold value: 0
> Option name: treble value: 0
> Screen resolution 1024 x 600 screenRotate 0 touch_swap setting 0
> Input device name: "C-Media Electronics Inc. USB Audio Device"
> Device is not a keyboard
> Input device name: "MOSART Semi. 2.4G Keyboard Mouse"
> Device is a keyboard
> end keyboard
> Input device name: "wch.cn USB2IIC_CTP_CONTROL"
> Input device name: "vc4-hdmi-1 HDMI Jack"
> Input device name: "vc4-hdmi-1"
> Input device name: "vc4-hdmi-0 HDMI Jack"
> Input device name: "vc4-hdmi-0"
> Input device name: "C-Media Electronics Inc. USB Audio Device"
> Input device name: "MOSART Semi. 2.4G Keyboard Mouse"
> Input device name: "MOSART Semi. 2.4G Keyboard Mouse System Control"
> [... 407 lines cut ...]
>     [ 0, r=  0.5000000] : halfband,  r=  0.5
>     [ 1, r=  0.2500000] : halfband,  r=  0.5
>     [ 2, r=  0.1250000] : halfband,  r=  0.5
>     [ 3, r=  0.1282462] : arbitrary, r=  1.02596998
> underrun resamplerate 0.128246 -0.009000

In sdrberry_settings.cfg is in hackrf section the items defined?
[hackrf]
...
thresholdDroppedFrames = "1"
thresholdUnderrun = "2"

Maybe you can share the hackrf section from sdrberry_settings.cfg

Mavik7zip, 2025-06-29:
i have changed samplerate 4000 to 8000 but same issue

```
[hackrf]
audiobuffersize = "4096"
bandwidth = "0"
dc = "0"
decimate = 5
drive = "50"
halfduplex = "1"
if-gain = 0
resample_rate = {"2.038080", "8000", "5"}
restart_tx = "1"
rf-gain = "10"
rf-gain-map = {"13m", "97", "17m", "0", "20m", "89", "2m", "83", "40m", "97",>
samplerate = 8000
samplerate_tx = 8000
span = 250
```

i have add thresholdDroppedFrames = "1", thresholdUnderrun = "2" I added this and it doesn't seem to have that lag anymore. I'll check better these days. 
thanks so much

paulh002, 2025-06-29:
The line in the log "Buffer queue 1236 Radio samples 8192 Audio Samples 1041 Passes 446 Queued Audio Samples 2048 droppedframes 0 underrun 18" shows the mismatch between audiosamplerate and SDR sample rate. The Queued Audiosamples should not grow to large. The thresholds manage the feedbackloop and adjust the resample rate to match the audiosample rate. If sdrberry changes the resample rate all the time you can adjust the thresholdDroppedFrames or thresholdUnderrun

Mavik7zip, 2025-06-30:
hello sorry for the log you are right I made a mistake in copying and pasting anyway the audio problem is what I show in the video of the first message of the issue. now I no longer have problems with delay or waterfall and I thought that this change would affect the audio buffer and maybe fix the problem of choppy audio in transmission but it is not so. I also tried to reselect the audio device as you told me and in this installation there is no pulseaudio but the problem continues to be there. bye

paulh002, 2025-06-30:
Ok, the choppy audio is the problem. I will check it. It can be an issue in the decimation (rx) / interpolation (tx) of the sample rate. The audio has to be up sampled to 8000 Khz sample rate. That is maybe not working well. You could check this by trying a low sample rate without decimation.

Mavik7zip, 2025-07-01:
hi I just did the test you told me and yes I can confirm that the problem is in the samplerate the only samplerate that works correctly is 4000khz with decimation between 2-32 instead the tx samplerate doesn't seem to change anything

Mavik7zip, 2025-07-01:
hi ok so it works correctly even with the samplerate at 8000khz, I wanted to ask you if it is normal that in the build version the changes to the lv_drv_conf.h file are excluded. thanks again and good work

paulh002, 2025-07-01:
Good! Yes that file contains some gui settings which for certain displays
and is locally updated. Then git will not change it.

paulh002, 2025-07-01:
I implemented this decimation for hackrf. It seems that hackrf is not
stable below 8000khz sample rate. But the pi4 is not fast enough to process
the complete 8mhz samples . So decimation solves this issue.