paulh002/sdrberry #68: Mouse control reversed with screen rotation

jcourter, 2025-06-16:
I bought an enclosure for the Touch 2 display that mounts the display oriented so that I need to use rotation="3" in the configuration file.  This gives me a correct display, but when I move the mouse, it's upside-down (i.e. left/right and up/down are reversed).  Is there a way to correct this?

Thanks!

paulh002, 2025-06-16:
I have to check, There is some rotation logic. But I have to test it.
Which housing are you using? Until now I did not find anything.

jcourter, 2025-06-16:
I am using the Smarti Case Touch Pro 3 case (https://smarticase.com/).  On first look, it appears that the code in mouse.cpp is only the swapping x and y axes, not inverting them when the display is mounted horizontally and upside-down.

paulh002, 2025-06-16:
I see, it is not the official touch 2 but some compatible type. But I have to check if I don't break anything for other type of displays. There are other people using rotation=3 and there it seems no issue. I will test with a official touch2 but I cannot remember I had this problem.

jcourter, 2025-06-16:
This is an official Pi Touch Display 2 (confusingly, Smarti Case called their enclosure for the original Touch display the "Touch 2" and the enclosure for the Pi Touch 2 display is called the "Touch Pro 3").  I _think_ the following changes (in bold) to mouse.cpp would correct this for my display.  I think a display requiring rotate="2" would probably need a similar adjustment.  I'm going to give it a try shortly and I'll let you know.

if (mouse_event.type == EV_REL && mouse_event.code == rel_1 )
			{
				if (IsScreenRotated() == 1)  **// got rid of the condition for IsScreenRotated()=="3" because it's doing the opposite of what I need right now**
					state.x = state.x + mouse_event.value * -1;
				else
					state.x = state.x + mouse_event.value;
				if (state.x >= screen_w)
					state.x = screen_w - 1;
				if (state.x < 0)
					state.x = 0;
				//printf("value %d, x %d\n", mouse_event.value, state.x);
				state.MouseActivity = true;
			}
			if (mouse_event.type == EV_REL && mouse_event.code == rel_2)
			{
				**if (IsScreenRotated() == 3)  // added this to invert the values
					state.y = state.y + mouse_event.value * -1;
				else**
					state.y = state.y + mouse_event.value;
				if (state.y >= screen_h)
					state.y = screen_h - 1;
				if (state.y < 0)
					state.y = 0;
				//printf("value %d, y %d\n", mouse_event.value, state.y);
				state.MouseActivity = true;	
			}

jcourter, 2025-06-16:
Confirmed, the change above fixes that.  I'm not sure if a display exists where the rotation would need to be set to 2, but I think it would probably need to have both axes inverted, in other words, I think both if statements should also have "|| IsScreenRotated() == 2" as a condition, but I don't have a way to test that.  Thanks for this awesome bit of software, I love it.

paulh002, 2025-06-16:
ok, will add it.
thanks