paulh002/sdrberry #72: can't get sdrberry running on pi4b

Urminme, 2025-06-27:
I've tried multiple times to get this going on my pi4b but it just never runs I always end up back at the terminal, the latest try I updated the pi os via update and upgrade, then cloned in the the repo listed on this page, after everything downloaded and installed it restarts but its showing tons of errors like radioberry doesnt exit or werror=incompatible-pointer-types and fatal: module radioberry not found.  I dont know if this is not compatible with the latest Raspberry or what but I just need some help.

paulh002, 2025-06-27:
Some repo's needed to be update. Also install script needed some updates due to pi OS updates. You can try again with a clean install.

Urminme, 2025-06-28:
So still not working, at one point it just freezes and disconnects from SSH, ive tried multiple times, not sure why its still not working but I dont know what else to try where it freezes is here

-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done

I never get to the point where it asks if I have a 16 or 25 like before, but I also restart and checked if soapysdr is there but it doesnt see it either, tried with sd card and usb neither work

paulh002, 2025-06-28:
Don't use a SD card it is to slow, cmake is downloading libraries that
takes some time. Or increase the timeout.

Urminme, 2025-06-28:
> Don't use a SD card it is to slow, cmake is downloading libraries that
> takes some time. Or increase the timeout.
> 
> Op za 28 jun. 2025 07:07 schreef Urminme ***@***.***>:
> […](#)

Ive tried with USB with same same issues

Urminme, 2025-06-28:
> I just installed it over here several times and no issue. The problem you describe is cmake downloading phase. You can try it manually. Go to sdrberry directory then to build and execute cmake 
> ..
> 
> ```
> cd sdrberry
> cd build
> cmake ..
> make -j4
> sudo make install
> ```

So if I make manually do I need to download all the other parts like soapy separately? I'll try again and see where that gets me I don't know why it wouldn't work for me only, I'm not sure what I'm doing wrong

Urminme, 2025-06-29:
> I just reinstalled for radioberry and it worked fine for mer. As alternative connect a keyboard to the pi and install directly from the commandline maybe the slowness is due to internet connection speed.

I finally was able to get it downloaded and it appears to be working, but im going to mess with it a little and make sure its all working

Urminme, 2025-06-30:
> To be able to transmit you need an USB audio adapter, the cheap ones for aliexpress are fine. Make sure you run the software in CLI mode, use raspi-config to switch to CLI mode. Set the radioberry sample rate to 96 Khz and try again. You can find my email on qrz.com. Please also check the wiki for help first. If you still have issues share the log of sdrberry.

I checked for you on QRZ but no email, I went to try and use the Pi today and now when I run the start command it just says "Segmentation fault", I've tried restarting but it's still not working again. I really dont want to re-flash another stick and go through the 45 minute process of downloading and setting it up again, anything you can recommend?

Urminme, 2025-07-02:
> Also check with dmesg to see if radioberry driver is loaded. You can also check with SoapyUtil if the SoapyRadioberry driver is available. You can check here for usage https://manpages.debian.org/testing/soapysdr-tools/SoapySDRUtil.1.en.html

So I have purchased a new Pi4b, new memory card, new usb audio adaptor and went through all the steps again and it downloaded find, seems to run fine, but still when I am on the SDR settings it wont let me select bandwidth, also the spectrum waterfall is all static and snow but its in red, which I havent seen on others with SDRberry, alos the IF Gain bar is red as well not blue, I tuned to the 20m wsjt frequency and can hear it through headphones plugged into the usb audio adaptor, but if I go to WSJT and set monitor nothing shows up, I've set it up with my home multiband antenna tuned to 20m. I have a log, but cant find the sdrberry settings 

[SDRberry log.txt](https://github.com/user-attachments/files/21011044/SDRberry.log.txt)

[SDRberry driver and soapySDR.txt](https://github.com/user-attachments/files/21011054/SDRberry.driver.and.soapySDR.txt)

Urminme, 2025-07-02:
> dmesg | grep Radioberry

this is what I get 

pi@sdrberry:~ $ dmesg | grep Radioberry
[    3.714988] Radioberry: registered correctly with major number 238
[    3.715016] Radioberry: device class registered correctly
[    3.716003] Radioberry char: device class created correctly
[    3.716132] Radioberry: The rx samples pin is mapped to IRQ: 43
[    3.716136] Radioberry: The rx sample state is currently: 0
[    3.716171] Radioberry: The interrupt request result is: 0
[    3.737255] Radioberry gateware file size: 335337

paulh002, 2025-07-02:
Bit strange, looks like gateware not loading. One thing do you start sdrberry as root? Sudo sdrberry?. Radioberry device is mostly only accessible as root.

Urminme, 2025-07-02:
> Bit strange, looks like gateware not loading. One thing do you start sdrberry as root? Sudo sdrberry?. Radioberry device is mostly only accessible as root.

Yes I always use Sudo sdrberry > sdrberry.log 2>&1 This has happened with both Rapberry Pi devices, only things thats not been changed is the radioberry device it self

Urminme, 2025-07-02:
These are screen shots from when its running on my system 

[image]
[image]

paulh002, 2025-07-02:
Something is wrong with the radioberry driver, did you activate SPI in raspi-config? That should not be done.

Urminme, 2025-07-02:
I don't think i did but I can double check and if so I will deactivate it and retry, also I never downloaded any firmware on the radioberry it self just plugged it in as i received it so Im not sure if the firmware for it is part of the sdrberry download or if I need to flash it seperately. thank you again for helping with this and sorry for making your life more complicated

Urminme, 2025-07-02:
> Something is wrong with the radioberry driver, did you activate SPI in raspi-config? That should not be done.

I made sure SPI is disabled and rebooted, no change, still no bandwith option or TX antenna, also below in the box that says setting and value its blank now. Spectrum is still static. Dmesg is still the same no change from previous

Urminme, 2025-07-02:
> Can you send the complete output of dmesg? I think we are missing something. I am sure the driver is not correct loading

This is a screen shot of the output, thats all it gives me unless im not entering the command correctly

[image]

paulh002, 2025-07-02:
Just dmesg not with grep, that filters the output

Urminme, 2025-07-02:
[SDRBERRY DMESG.txt](https://github.com/user-attachments/files/21024411/SDRBERRY.DMESG.txt)

Urminme, 2025-07-02:
I wonder whats casuing these issues, im stumped, I am unfortunately not a Linux or Pi expert

paulh002, 2025-07-02:
Assuming the driver is ok, can you switch the sample rate to 96 khz? And make sure the noise suppression is off. It is switch on in the picture

paulh002, 2025-07-02:
Only the Rx samplerate

paulh002, 2025-07-02:
You can test the radioberry with SoapySDRUtil. 
Execute following command from the commandline:
```
SoapySDRUtil --args="driver=radioberry" --rate=96000 --direction=RX
```

The result should look like this:

[image]

Urminme, 2025-07-06:
> That looks ok, unfortunately I cannot replicate your issue. You first have to check if the radioberry produces the correct data. My advice is to install the firmware and use a pc sdr software to check if the radioberry works. You can use the installation script in Radioberry/SBC/rpi-4/releases/dev/radioberry_install.sh this installs the driver and the firmware if you start the firmware you connect usung a pc. See the wiki of radioberry.

So I replaced the Radioberry I purchased and used the new raspberry Pi4B, followed all the same steps and it does seems to work better now, I can hear the radios working, the waterfall is not just red static or snow now, but the bandwidth option in SDR settings is still blank, doesnt give me any options and same with antenna just RX antenna only, the other thing I noticed is if I try and use WSJT it just freezes then crashes when I press monitor and after crashing says "aborted",. I dont know what other variables to change Ive replaced every piece of equipment Im using for this project

paulh002, 2025-07-07:
The options Bandwidth and RX antenna are not used for radioberry. (for hackrf and sdrplay). The issue with wsjtx is my mistake. I fixed the issue you can update sdrberry with following steps:

```
cd sdrberry
git pull
cd build
cmake ..
make -j4
sudo make install
```

Urminme, 2025-07-11:
> The options Bandwidth and RX antenna are not used for radioberry. (for hackrf and sdrplay). The issue with wsjtx is my mistake. I fixed the issue you can update sdrberry with following steps:
> 
> ```
> cd sdrberry
> git pull
> cd build
> cmake ..
> make -j4
> sudo make install
> ```

Got it that makes sense, so don't worry about the bandwidth. I did follow the instructions you provided and the download seemed to go just fine but when I try using Wsjt it just freezes and says Aborted still. Not sure if I need to enter these commands one at a time, but I normally just copy and paste.

RobertSwirsky, 2025-07-16:
Mine won't build either. Haven't spent much time debugging it, but I've tried it twice on clean installs on an RPi4 (the version that best supports radioberry) and there never is an sdrberry executable created in /usr/local/bin 

I'm doing it from user "pi" -- the only user on the system.

I tried running install.sh both as root and as pi

RobertSwirsky, 2025-07-16:
Thanks! I Added CatTcpServer.cpp to line 25 of CMakeList.txt and the build succeeded and it starts up.

paulh002, 2025-07-18:
I updated CMakeList.txt it should compile again.

closed 2025-07-23 by paulh002