paulh002/sdrberry #77: VFO display update (CAT)

PA1ASB, 2025-09-28:
Changes made to the VFO frequencies through FT-891 CAT command FA / FB are not updated on the monitor. The frequencies returned by SDRBerry reflect the frequencies set. I tried to figure out how I can force an update on the screen but, not being a programmer, I can't find out where to look. Anybody who can help me here? Thanks in advance! 

Menno  PA1ASB

paulh002, 2025-09-28:
I will have a look

paulh002, 2025-09-28:
I have checked the issue, but cannot replicate the issue. The GUI updates correctly
Please update to the latest version, if the issue persist please show a screenshot and the steps executed to replicate the issue.

73,
Paul

PA1ASB, 2025-09-28:
Hello Paul,

Thanks for the swift reply. I have made a clean installation according to the instructions listed here but same result. If I change, for instance, band or volume or whatever command possible I can see that the GUI is updated instantly so there is obviously something fishy here... The only thing I have changed in the source is the colordepth: I'm using a Raspberry Touchscreen 2 which is 32 bit colordepth but requires this parameter to be set to 16 (same issue as reported by Hellberg78) but I can hardly believe that that would cause this problem. 
In any case, I will make some screenshots tomorrow of the ongoing communication between my controller and the Pi; have to make some changes in the code to visualize that neatly, but I call it a day for now. 

Again thanks for your prompt action.
72
Menno  PA1ASB

PA1ASB, 2025-09-29:
Paul,

Attached please find a copy of the logfile. I have reduced its length to keep it somewhat compact, don't believe that anything related to CAT comms is left out. For clarity: I'm using CAT over serial, for now is a simple TTL to USB converter which shows up in the Pi as ttyUSB0 so that is the port that I'm referencing. I reduced baudrate to 19200 though at much higher rates it still works fine. I also made a screenshot of de serial terminal in Arduino (I use a Teensy 4.1 to communicate with the Pi) and this seems a little strange. First off, I would expect a reply to be sent after changing VFO frequency with 'AF....;' but nothing is returned. 
However, when I send 'AF;'  the Pi returns the frequency that I actually have set although that is not reflected on the screen. Also, when I change band using the touchscreen buttons, the lower frequency of the band is displayed and the displayed frequency for that VFO is actually returned through the CAT interface as well. However, when I then change the frequency setpoint and send it back over CAT, nothing actually changes though it's reported back that it actually hás changed. 
When I look into the logfile, SDRBerry apparently does not send the altered VFO data to Soapy so that would explain why there is no update. Then still it remains weird that radioStatus.FA / FB are actually changed... In any case: there is a discrepancy between what is reported by CAT and what is displayed (and the VFO is actually tuned to).

[sdrberry.log](https://github.com/user-attachments/files/22591414/sdrberry.log)

[image]

72
Menno PA1ASB

paulh002, 2025-09-29:
Ok, now I have some more information. Will try to reproduce. When you change the frequency sdrberry does not change the frequency of the SDR when the frequency is in range of the sampled bandwidth. It down converts the requested frequency to baseband. Only when you exceed the sampled bandwidth it will change the requested frequency of the SDR. 

You can also sen FT commands instead of FA/FB this is a FrequencyTune command which sends frequency steps (delta's) this I use for sending updates from an optical rotary controler.
You can check the ESP32 Arduino code in gui.cpp as example.

https://github.com/paulh002/ESP32Controlerv2

I will check the FA / FB commands why they don't update the gui.

PA1ASB, 2025-09-29:
Paul,

I have tested the 'FT' command and that actually updates both the setpoint AND the display. Although this would make it possible to move around all bands it makes the software a little more complicated to quickly jump from one to another frequency - or at least it could take quite a number of commands to be sent to and fro. The reason why I make a point of this is that the 5" (or even 7") displays are too small for me to read and/or use the touch buttons on-screen to control the rig. I therefore want "massive" controls / buttons / rotary encoders and the like. So band switching and VFO setting etc. is all handled outside of the Pi just as settings for all bands used, including memories and the likes. SDRBerry 's GUI thus serves only as a display, not as a control system and I actually am looking to remove most of these controls from the screen; I believe that is quite feasible.

Regards and 72
Menno PA1ASB

PA1ASB, 2025-09-29:
OK, installed the build version and that one seems to be working.
Thanks for your effort and very, very rapid responses.

Best regards and 72
Menno PA1ASB